tinygo-org/tinygo #3: Flawed a fortiori argument

isaachier, 2018-08-19:
I'll admit this isn't a bug, just responding to this sentence in the README:

> My original reasoning was: if Python can run on microcontrollers, then certainly Go should be able to and run on even lower level micros.

The problem here is the last part: "even lower level micros." Now, consider this valid argument from Wikipedia:

> If driving 10 mph over the speed limit is punishable by a fine of $50, it can be inferred a fortiori that driving 20 mph over the speed limit is also punishable by a fine of at least $50.
(https://en.wikipedia.org/wiki/Argumentum_a_fortiori)

If I were to say the same thing, but finish with "20 mph . . . at least $100," that would be an invalid deduction. My suggestion here is to correct the README to say ". . . then certainly Go should be able to run on microcontrollers and potentially lower level micros."

aykevl, 2018-08-20:
I guess you are technically correct about the wording, but the conclusion is in fact true. Python is an interpreted language so requires a relatively big program to run it (MicroPython is at the very least 70kB in size). This is very different for Go: as it is a compiled program all unused features can be left out meaning it can run on very small microcontrollers. So it is reasonable to assume it will require less resources than MicroPython (and in fact I've already done so).

I could change the sentence to something like this, if you'd prefer ;)

> if Python (an interpreted language) can run on microcontrollers, then certainly Go (a compiled language) should be able to and run on even lower level micros.

isaachier, 2018-08-20:
How about something like: "if python can run with an entire interpreter on a microcontroller, certainly Go can run compiled programs that use less memory/disk." whatever you decide, I'm glad I posted this issue because I now understand what you meant more clearly than from the original argument I saw in the readme.

isaachier, 2018-08-20:
Seeing as this is more of a nitpick anyway, closing the issue.

closed 2018-08-20 by isaachier