tinygo-org/tinygo #4: Initial setup confusion

deadprogram, 2018-09-12:
Hi @aykevl this project looks really cool.

So I am trying to get my environment setup correctly. Seems like I should be using your fork of llvm. https://github.com/aykevl/llvm since it has several important looking commits. The Makefile has it located in the GOPATH https://github.com/aykevl/tinygo/blob/master/Makefile#L9 would this be where I should check that repo out? Also, where should the llvm-build and llvm-inst directories be set to?

Thanks for helping me get started with tinygo.

aykevl, 2018-09-12:
Happy to see you're interested :D

Did you read the 'Building' section of the README? (https://github.com/aykevl/tinygo#building). It should answer most of your questions. If not, just ask again.

> Seems like I should be using your fork of llvm. https://github.com/aykevl/llvm since it has several important looking commits.

Yes, I needed to add a few Go APIs for coroutines and low-level stuff, so that's why I made the fork. All patches have been submitted upstream and most have been accepted so this fork should be unnecessary in the future. Additionally, I'd like to transition to a system-installed version of LLVM at some point to avoid the need for manually building LLVM so `go get` just works.

deadprogram, 2018-09-12:
Thanks for the reply. The sentence

```
To continue, you'll need to build LLVM for Go.
```

Is all it says in the Readme. That is where my questions come from. Perhaps a specific list of locations for the destination directories use for the llvm build?

aykevl, 2018-09-12:
Ah, of course. I forget to mention that. Building is really easy: simply run `./build.sh` in $GOPATH/github.com/aykevl/llvm/bindings/go.

deadprogram, 2018-09-12:
OK got it.

To summarize:

1. go get of tinygo also pulls the patched fork of llvm
2. modify the llvm fork bindings/go/build.sh to allow AVR llvm backend
3. run bindings/go/build.sh

I am about 77% done running step 3....

deadprogram, 2018-09-12:
OK llvm finished building. However `make run-test` fails:

```
$ make run-test
go build -o build/tgo -i .
/home/ron/.gvm/pkgsets/go1.11/global/src/github.com/aykevl/llvm/bindings/go/llvm/workdir/llvm_build/bin/llvm-as -o buil
d/runtime.bc src/runtime/runtime.ll
/home/ron/.gvm/pkgsets/go1.11/global/src/github.com/aykevl/llvm/bindings/go/llvm/workdir/llvm_build/bin/llvm-link -o bu
ild/runtime-unix-combined.bc build/runtime.bc
./build/tgo  -runtime build/runtime-unix-combined.bc -o build/test.o examples/test
clang -o build/test build/test.o
make: clang: Command not found
Makefile:129: recipe for target 'build/test' failed
make: *** [build/test] Error 127
rm build/test.o
```

I then tried to `git clone https://github.com/llvm-mirror/clang` from within the `/src/github.com/aykevl/llvm\tools` dir, and re-run the build.sh. Build ran, but same error running `make run-test`

deadprogram, 2018-09-12:
I see I was taking the instructions for llvm to literally, and just installing a recent version of clang in my normal package manager worked:

```
$ make run-test
./build/tgo  -runtime build/runtime-unix-combined.bc -o build/test.o examples/test
clang -o build/test build/test.o
size build/test
   text    data     bss     dec     hex filename
   7437     768       8    8213    2015 build/test
./build/test
Hello world from Go!
The answer is: 42
...
```

OK on to the AVR! Thanks for the help @aykevl

closed 2018-09-12 by deadprogram

aykevl, 2018-09-12:
Great to see that it works!
A warning, though: AVR support is *highly* experimental in LLVM and it only works for extremely trivial programs.

> I see I was taking the instructions for llvm to literally, and just installing a recent version of clang in my normal package manager worked:

Yes, it just needs to compile those files. An old version works just fine.