tinygo-org/tinygo #5: Unable to build blinky example for Arduino

deadprogram, 2018-09-12:
When trying to build/run the blinky2 example with TARGET=arduino I get the following errors:

```
$ make run-blinky2 TARGET=arduino                                                                                      
./build/tgo -target arduino -runtime build/runtime-arduino-combined.bc -o build/blinky2.o examples/blinky2
src/machine/machine_avr.go:38:9: PORTD not declared by package avr
src/machine/machine_avr.go:40:9: PORTB not declared by package avr
src/machine/machine_avr.go:44:9: PORTB not declared by package avr
src/machine/machine_avr.go:46:9: PORTB not declared by package avr
src/machine/machine_avr.go:22:9: DDRD not declared by package avr
src/machine/machine_avr.go:24:9: DDRB not declared by package avr
src/machine/machine_avr.go:28:9: DDRD not declared by package avr
src/machine/machine_avr.go:30:9: DDRB not declared by package avr
src/runtime/runtime_avr.go:76:7: WDTCSR not declared by package avr
src/runtime/runtime_avr.go:76:21: WDTCSR_WDCE not declared by package avr
src/runtime/runtime_avr.go:76:39: WDTCSR_WDE not declared by package avr
src/runtime/runtime_avr.go:78:20: WDTCSR_WDIE not declared by package avr
src/runtime/runtime_avr.go:78:38: RegValue not declared by package avr
src/runtime/runtime_avr.go:78:7: WDTCSR not declared by package avr
src/runtime/runtime_avr.go:84:29: SMCR_SE not declared by package avr
src/runtime/runtime_avr.go:84:7: SMCR not declared by package avr
src/runtime/runtime_avr.go:90:7: SMCR not declared by package avr
src/runtime/runtime_avr.go:46:12: UCSR0A not declared by package avr
src/runtime/runtime_avr.go:46:25: UCSR0A_UDRE0 not declared by package avr
src/runtime/runtime_avr.go:49:18: RegValue not declared by package avr
src/runtime/runtime_avr.go:49:7: UDR0 not declared by package avr
src/runtime/runtime_avr.go:39:7: UBRR0H not declared by package avr
src/runtime/runtime_avr.go:40:7: UBRR0L not declared by package avr
src/runtime/runtime_avr.go:41:20: UCSR0B_RXEN0 not declared by package avr
src/runtime/runtime_avr.go:41:39: UCSR0B_TXEN0 not declared by package avr
src/runtime/runtime_avr.go:41:7: UCSR0B not declared by package avr
src/runtime/runtime_avr.go:42:20: UCSR0C_UCSZ0 not declared by package avr
src/runtime/runtime_avr.go:42:7: UCSR0C not declared by package avr
src/examples/blinky2/blinky2.go:33:30: LED2 not declared by package machine
error: src/runtime/runtime_avr.go:76:7: WDTCSR not declared by package avr
Makefile:99: recipe for target 'build/blinky2.o' failed
make: *** [build/blinky2.o] Error 1
```

deadprogram, 2018-09-12:
Figured out that you need to run `make gen-device-avr` to generate the needed device files. 

Also added a const LED2=2 in machine/machine_avr.go

Now getting the following:
```
$ make run-blinky2 TARGET=arduino
go build -o build/tgo -i .
/home/ron/.gvm/pkgsets/go1.11/global/src/github.com/aykevl/llvm/bindings/go/llvm/workdir/llvm_build/bin/llvm-as -o build/runtime.bc src/runtime/ru
ntime.ll
/home/ron/.gvm/pkgsets/go1.11/global/src/github.com/aykevl/llvm/bindings/go/llvm/workdir/llvm_build/bin/llvm-link -o build/runtime-arduino-combine
d.bc build/runtime.bc
./build/tgo -target arduino -runtime build/runtime-arduino-combined.bc -o build/blinky2.o examples/blinky2
tgo: /home/ron/.gvm/pkgsets/go1.11/global/src/github.com/aykevl/llvm/lib/CodeGen/SelectionDAG/SelectionDAGBuilder.cpp:9039: void llvm::SelectionDA
GISel::LowerArguments(const llvm::Function&): Assertion `InVals.size() == Ins.size() && "LowerFormalArguments didn't emit the correct number of va
lues!"' failed.
SIGABRT: abort
PC=0x7f3d1575e428 m=5 sigcode=18446744073709551610

goroutine 0 [idle]:
runtime: unknown pc 0x7f3d1575e428
stack: frame={sp:0x7f3d0ef9b5c8, fp:0x0} stack=[0x7f3d0e79d2a8,0x7f3d0ef9cea8)
....
```

aykevl, 2018-09-12:
Sadly, blinky2 is too complex for AVR at the moment. blinky1 should work, however.

I'd like to improve this, but it hasn't been done yet. It may require changes to LLVM.

> Figured out that you need to run make gen-device-avr to generate the needed device files.

Yes, that is required. I should improve the error messages for that, and probably add it to the README as well.

deadprogram, 2018-09-12:
I modded the Makefile target to build blinky1 instead of blinky2. Now here a different error:

```
$ make run-blinky1 TARGET=arduino
go build -o build/tgo -i .
/home/ron/.gvm/pkgsets/go1.11/global/src/github.com/aykevl/llvm/bindings/go/llvm/workdir/llvm_build/bin/llvm-as -o build/runtime.bc src/runtime/runtime.ll
/home/ron/.gvm/pkgsets/go1.11/global/src/github.com/aykevl/llvm/bindings/go/llvm/workdir/llvm_build/bin/llvm-link -o build/runtime-arduino-combined.bc build/runtime.bc
./build/tgo -target arduino -runtime build/runtime-arduino-combined.bc -o build/blinky1.o examples/blinky1
avr-as -mmcu=atmega328p -o build/avr.o avr.S
avr-ld -T avr.ld --gc-sections -o build/blinky1 build/blinky1.o build/avr.o
avr-size build/blinky1
   text    data     bss     dec     hex filename
    380       0     520     900     384 build/blinky1
./build/blinky1
./build/blinky1: 1: ./build/blinky1: Syntax error: "(" unexpected
Makefile:64: recipe for target 'run-blinky1' failed
make: *** [run-blinky1] Error 2
rm build/blinky1.o build/avr.o
```

aykevl, 2018-09-12:
That's because the `run-*` commands execute on the host. It is trying to execute an AVR binary on your host.

This should work:

    make flash-blinky1 TARGET=arduino

I have updated the README with this example.

deadprogram, 2018-09-12:
That worked, I now have a blinking Arduino Uno. Thanks for the help!

closed 2018-09-12 by deadprogram