tinygo-org/tinygo #10: How to use println on AVR

deadprogram, 2018-09-16:
I am trying to build the following code for AVR:

```go
package main

import (
	"time"
)

func main() {
	for {
		println("Hello world from Go!")
		time.Sleep(time.Millisecond * 1000)
	}
}
```

However it give the following on build:

```
$ make flash-serial TARGET=arduino
./build/tgo build -target arduino -o build/serial.elf examples/serial
tgo: /home/ron/.gvm/pkgsets/go1.11/global/src/github.com/aykevl/llvm/lib/CodeGen/SelectionDAG/SelectionDAGBuilder.cpp:$
039: void llvm::SelectionDAGISel::LowerArguments(const llvm::Function&): Assertion `InVals.size() == Ins.size() && "Lo$
erFormalArguments didn't emit the correct number of values!"' failed.
```

Seems like perhaps the `putchar()` functionality that is in https://github.com/aykevl/tinygo/blob/master/src/runtime/runtime_avr.go#L43 is not hooked up yet?

I thought the runtime would call that itself from just using the `println()` function, but did not work as expected.

deadprogram, 2018-09-16:
To elaborate, what I expected to happen was that the Arduino would start sending the string `""Hello world from Go!""` on the serial port to the connected host computer.

aykevl, 2018-09-16:
`println()` is implemented, and it is implemented in a platform-agnostic way:
* The builtin function is mapped to `runtime.print*` functions [here](https://github.com/aykevl/tinygo/blob/35fe24981f6cb177f8631ff40eeb6ce499800229/compiler.go#L1649).
* The `runtime.print*` functions (for every supported type) are implemented in [src/runtime/print.go](https://github.com/aykevl/tinygo/blob/master/src/runtime/print.go).

This code works for me. Can you try it?

```go
package main

import "time"

func main() {
	for {
		println("hello world!")
		time.Sleep(time.Second)
	}
}
```

When I try to compile your code, it compiles and flashes, but it prints garbage to the serial console (but the same garbage every second, so the code is otherwise working). There is a small difference in the generated assembly, which makes no sense to me. I'm still debugging it.

deadprogram, 2018-09-16:
This is the command I used:

```
make flash-serial TARGET=arduino
```

Which did not compile with the errors in the original issue.

aykevl, 2018-09-16:
I'm not sure why the original source above does not compile (it compiles for me), but I think I know why one source runs and the other doesn't: it only works sometimes by accident, and I got very lucky. Here is my theory. The AVR uses a Harvard-style architecture with a different address space for RAM and flash. For the AVR, this means a given pointer value may point to flash and to RAM, where of course only one is valid. TinyGo has not implemented any support for global constants (or even initialized global variables, only global zero-initialized variables) so when it tried to read the string data, it just prints whatever it finds at the RAM location while it should have loaded the data from flash. Apparently I got really lucky and the bootloader in my Arduino placed the string data at the RAM address that matches the flash address it should have been read from, so that's why my `println("hello world!")` happened to work, more-or-less (until the next power cycle).

tl;dr: global constants like strings haven't been implemented yet so cannot be relied upon. I don't know why it fails to compile but it certainly doesn't work at runtime, yet. I'll revisit this issue after global constants have been implemented.

aykevl, 2018-09-16:
I have pushed two commits that should improve (but not entirely fix) global constants, simply by making them non-constant as a temporary fix (determining which pointers point to flash and which to RAM is non-trivial in the general case or even impossible).

Can you test the first piece of code again and see whether anything has changed?

Note: global variables haven't been fixed entirely: the first byte is corrupted for some reason (printing "Dello world!").

deadprogram, 2018-09-16:
Pulled latest changes from master. Now I get:

```
$ make flash-serial TARGET=arduino
go build -o build/tgo -i .
./build/tgo build -target arduino -o build/serial.elf examples/serial
/usr/lib/gcc/avr/4.9.2/../../../avr/bin/ld: cannot find -lm
/usr/lib/gcc/avr/4.9.2/../../../avr/bin/ld: cannot find -lc
collect2: error: ld returned 1 exit status
error: exit status 1
Makefile:75: recipe for target 'build/serial.elf' failed
```

deadprogram, 2018-09-16:
I did not have `avr-libc` installed. Now compiles... output via screen command:

```
Dcow�lM�Dcow�lM�Dcow�lM�Dcow�lM�Dcow�lM
```

So that seems like progress!

deadprogram, 2018-09-16:
`screen /dev/ttyACM0 115200`

aykevl, 2018-09-16:
Have you maybe merged the PWM branch? I just discovered that it breaks UART, see https://github.com/aykevl/tinygo/pull/8#discussion_r217921373 for details.

deadprogram, 2018-09-16:
You are absolutely correct. Now I have the same thing as you see, the first char is mangled, but everything else is working.

Next up, any thoughts about how to best implement `io.reader` to getchar on the UART?

aykevl, 2018-09-16:
You're really going full steam ahead! :D

I have written a few of my thoughts in the wiki: https://github.com/aykevl/tinygo/wiki/Machine-API
If this UART is implemented, I think it is best to import it from the runtime package to avoid duplicated code. When done properly, this should not result in a code size increase. `putchar()` can then be implemented like this:

```go
func putchar(c byte) {
    machine.UART0.WriteByte(c)
}
```

Closing this issue as the original issue is resolved.

closed 2018-09-16 by aykevl