tinygo-org/tinygo #11: Reduce memory usage of slices

lpereira, 2018-09-16:
Since slices are copied all over the place, it might be a good idea to, instead of having a struct with `{ptr, len, cap}`, have one with `{ptr, len}` only, and calculate `cap` on demand. Aligning `len` to the next power of two would work, but also waste some bytes as the slice grows. Maybe a different sequence, like the one Python has for lists, would be better?

aykevl, 2018-09-16:
I don't think it is actually possible to calculate `cap()` on demand, because it can be set to an arbitrary value: consider `slice[:5:7]`, which sets `len` to 5 and `cap` to 7 no matter what they were before.

In general, I think the compiler will be able to optimize `cap` away in trivial cases where it is never used (no `append` or `cap` call). This will probably improve once I get around to splitting the slice struct into independent variables in function calls etc. which LLVM is better at optimizing away.

deadprogram, 2019-01-11:
Is this issue still relevant given all of the internal changes to TinyGo, or should it be closed?

aykevl, 2019-01-13:
I don't think there is anything I can do here. I don't see a way to perform this optimization while still adhering to the Go language specification. Additionally, slices in local variables and function parameters are easily optimized by LLVM: function parameters are split up in #20 and local variables are split up by LLVM. Unnecessary copies are usually avoided by dead code elimination passes and the deadargelim pass. This means that the `cap` part of a slice is often already eliminated.

If there is something I'm missing I'd like to know about it. For now, I'm going to close the issue.

closed 2019-01-13 by aykevl