tinygo-org/tinygo #19: Help needed in debugging AVR compilation errors

deadprogram, 2018-09-22:
I am working on the I2C interface for the AVR. The WIP branch is located here:

https://github.com/hybridgroup/tinygo/commit/f3c8e288413c2d89288875d96043ce1617db7c9e

My problem is that the compile works with the code I just committed, but if I uncomment this line https://github.com/hybridgroup/tinygo/blob/feature/avr-i2c/src/examples/i2clcd/i2clcd.go#L52 in my sample, the compile fails with:

```
$ make flash-i2clcd TARGET=arduino                                                                                     
./build/tgo build -target arduino -size=short -o build/i2clcd.elf examples/i2clcd
ROLW unimplemented
UNREACHABLE executed at /home/ron/.gvm/pkgsets/go1.11/global/src/github.com/aykevl/llvm/lib/Target/AVR/AVRExpandPseudo$
nsts.cpp:1316!
SIGABRT: abort
PC=0x7f7ec3df0428 m=4 sigcode=18446744073709551610
```

If I uncomment the following line instead https://github.com/hybridgroup/tinygo/blob/feature/avr-i2c/src/examples/i2clcd/i2clcd.go#L53

I get this error:

```
$ make flash-i2clcd TARGET=arduino
go build -o build/tgo -i .
./build/tgo build -target arduino -size=short -o build/i2clcd.elf examples/i2clcd
tgo: /home/ron/.gvm/pkgsets/go1.11/global/src/github.com/aykevl/llvm/lib/CodeGen/SelectionDAG/SelectionDAGBuilder.cpp:9
039: void llvm::SelectionDAGISel::LowerArguments(const llvm::Function&): Assertion `InVals.size() == Ins.size() && "Low
erFormalArguments didn't emit the correct number of values!"' failed.
SIGABRT: abort
PC=0x7f633a6f8428 m=5 sigcode=18446744073709551610
```

Perhaps this has to do with not yet implemented features on AVR?

aykevl, 2018-09-22:
I have seen the second error more often. It is due to a limitation in the AVR backend (it cannot deal with struct parameters, and `[]byte` is really a `{ptr, len, cap}` struct). See #20 for details. A temporary (ugly) workaround is to make the inliner more likely to run by increasing the value of the third parameter to `Optmize(opt, size, inliner)` in [main.go](https://github.com/aykevl/tinygo/blob/master/main.go#L47).

I haven't seen the first error before but it also looks like an unimplemented feature.

aykevl, 2018-09-23:
You could try again now. The second problem is probably fixed. The first problem should really be fixed in upstream LLVM, but might also be fixed as a side effect (who knows).

A trick that might fix the first problem (again, a dirty workaround) is to sleep the same time everywhere. The optimizer will recognize the pattern and do the expensive `int64` calculation at compile time.

aykevl, 2018-09-23:
Ah now I see. The trouble with the second uncommented line might be related to the memory allocator, which I have never tested on AVR (as it wasn't able to run any significant programs).

deadprogram, 2018-09-24:
You were correct that changing all of the calls to `time.Sleep()` to use the same value let the compiler run without errors and without changing the value of the inliner param to `Optimize()`. That is probably not a practical long-term solution, of course.

The latest version of master lets the compiler run using various different time values like my original code, with the inline param set to the lower value `50`.