tinygo-org/tinygo #27: BBC:MIcrobit support

deadprogram, 2018-09-28:
I am experimenting with trying to get any code working at all on the BBC:Microbit: https://github.com/hybridgroup/tinygo/commit/7e5afe34020e408f4b7bf064dafb448067b2ab3f 

I am able to compile and load code on to the Microbit, but nothing seems to happen. I get the feeling my code is not actually executing... any thoughts?

BTW here is how you install J-Link on the Microbit: https://www.segger.com/products/debug-probes/j-link/models/other-j-links/bbc-microbit-j-link-upgrade/

aykevl, 2018-09-28:
I see two problems right away:
1. The CPU isn't set correctly. This is a Cortex-M0 instead of a Cortex-M4, so the LLVM target is `armv6m-none-eabi` and the linker (actually just GCC) must be passed `-mcpu=cortex-m0` instead of `-mcpu=cortex-m4`. This will crash the CPU immediately.
2. As far as I know, all micro:bit chips have 16K RAM instead of 64K (check it with the chip number, if it says QFAA it's 16K, if it says QFAC it is 32K). I don't think this is the issue, but you could try changing it.

deadprogram, 2018-09-28:
I had already found and changed the first one. When I change that, I receive the following compilation error:

```
$ make flash-serial TARGET=microbit
go build -o build/tgo -i .
./build/tgo build -target microbit -size=short -o build/serial.elf examples/serial
/tmp/tinygo896560428/main.o: In function `runtime.ticks':
/home/ron/.gvm/pkgsets/go1.11/global/src/github.com/aykevl/tinygo/src/runtime/runtime_nrf.go:79: undefined reference to `__umodsi3'
collect2: error: ld returned 1 exit status
error: exit status 1
Makefile:97: recipe for target 'build/serial.elf' failed
make: *** [build/serial.elf] Error 1
```

aykevl, 2018-09-28:
Also, have you ever tried remote debugging? There is some (limited) support for debug symbols in the compiler, enough to print a stack trace. You'll probably see it hits a HardFault. Debugging a hard fault is difficult on the cortex-m0, but there is [a trick](https://devzone.nordicsemi.com/f/nordic-q-a/7575/how-to-trace-back-hard-fault-handler/27005#27005) to get the previous program counter, which can then be converted to a real function by inspecting the result of `arm-none-eabi-objdump -d build/<example>.elf`.

aykevl, 2018-09-28:
Ah, you probably need to remove `-nostdlib`, see 0d8a7e16662d236a803d4c9a43b49f25067969df for a similar change.

aykevl, 2018-09-28:
Actually, I was yesterday considering buying a micro:bit as well as it's quite popular, will probably do that soon.

deadprogram, 2018-09-28:
Removing `-nostdlib` results in:

```
$ make flash-serial TARGET=microbit
go build -o build/tgo -i .
./build/tgo build -target microbit -size=short -o build/serial.elf examples/serial
/usr/lib/gcc/arm-none-eabi/4.9.3/armv6-m/libgcc.a(unwind-arm.o): In function `get_eit_entry':
/build/gcc-arm-none-eabi-DRRnLA/gcc-arm-none-eabi-4.9.3+svn231177/build/arm-none-eabi/armv6-m/libgcc/../../../../src/libgcc/unwind-arm-common.inc:274: undefined reference to `__exidx_start'
/build/gcc-arm-none-eabi-DRRnLA/gcc-arm-none-eabi-4.9.3+svn231177/build/arm-none-eabi/armv6-m/libgcc/../../../../src/libgcc/unwind-arm-common.inc:274: undefined reference to `__exidx_end'
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/armv6-m/libc.a(lib_a-abort.o): In function `abort':
/build/newlib-5zwpxE/newlib-2.2.0+git20150830.5a3d536/build/arm-none-eabi/armv6-m/newlib/libc/stdlib/../../../../../../newlib/libc/stdlib/abort.c:63: undefined reference to `_exit'
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/armv6-m/libc.a(lib_a-signalr.o): In function `_kill_r':
/build/newlib-5zwpxE/newlib-2.2.0+git20150830.5a3d536/build/arm-none-eabi/armv6-m/newlib/libc/reent/../../../../../../newlib/libc/reent/signalr.c:61: undefined reference to `_kill'
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/armv6-m/libc.a(lib_a-signalr.o): In function `_getpid_r':
/build/newlib-5zwpxE/newlib-2.2.0+git20150830.5a3d536/build/arm-none-eabi/armv6-m/newlib/libc/reent/../../../../../../newlib/libc/reent/signalr.c:97: undefined reference to `_getpid'
collect2: error: ld returned 1 exit status
error: exit status 1
Makefile:97: recipe for target 'build/serial.elf' failed
make: *** [build/serial.elf] Error 1
```

deadprogram, 2018-09-28:
Yes, the microbit is very common, hence why I started playing with this.

Also a chance to have 2 machines in the same family to work out side-by-side compilation, toolchain difference, etc.

aykevl, 2018-09-28:
I'm suspecting a bit more linker magic is required, probably to remove the personality function.

Will need to investigate that.

deadprogram, 2018-09-28:
Some progress on compiling/linking. Still does not work yet.

https://github.com/hybridgroup/tinygo/commit/1397fca4da60b217afbeabd3607727634b85b3eb

aykevl, 2018-09-29:
I'm looking into it. The weird thing is, as soon as I add `-lgcc` to GCC (after the object files - this matters), it suddenly requires all kinds of symbols, even though they should be discarded by the linker (`-Wl,--gc-sections`). I wonder why they're not being discarded.

aykevl, 2018-09-29:
Fixed a bug in 5bf058a0a61df64f6b150443926fd1456dafb432 which caused the undefined reference. You should now be able to compile for the nrf51 basically by copying pca10040.json and modifying the architecture, chip number and build tags - I am able to compile and run the blinky2 example on a PCA10040 when I switch to `armv6m-none-eabi` and `-mcpu=cortex-m0`.

aykevl, 2018-09-29:
For the future, I should look into adopting [compiler-rt](https://compiler-rt.llvm.org/) to depend less on the oddities of GCC.

aykevl, 2018-09-29:
> `"llvm-target": "armv6k-none-eabi",`

typo in `armv6k`?
Otherwise the file looks good.

deadprogram, 2018-09-29:
That was because I had originally received an error about needing that architecture for some instruction when I used  `armv6m-none-eabi` as you originally suggested. I cannot reproduce that error now, of course. Switched in the correct value, but still nothing happening.

aykevl, 2018-09-30:
I managed to get a simple "hello world" to run in an [emulator I once wrote](https://github.com/aykevl/emculator). This is the diff I used:

```diff
diff --git a/targets/nrf52.ld b/targets/nrf52.ld
index a388cf7..442ae1b 100644
--- a/targets/nrf52.ld
+++ b/targets/nrf52.ld
@@ -2,7 +2,7 @@
 MEMORY
 {
     FLASH_TEXT (rw) : ORIGIN = 0x00000000, LENGTH = 256K /* .text */
-    RAM (xrw)       : ORIGIN = 0x20000000, LENGTH = 64K
+    RAM (xrw)       : ORIGIN = 0x20000000, LENGTH = 16K
 }
 
 _stack_size = 2K;
diff --git a/targets/pca10040.json b/targets/pca10040.json
index 8aec981..74b507e 100644
--- a/targets/pca10040.json
+++ b/targets/pca10040.json
@@ -1,8 +1,8 @@
 {
-       "llvm-target": "armv7em-none-eabi",
+       "llvm-target": "armv6m-none-eabi",
        "build-tags": ["pca10040", "nrf52832", "nrf52", "nrf", "arm", "js", "wasm"],
        "linker": "arm-none-eabi-gcc",
-       "pre-link-args": ["-nostdlib", "-nostartfiles", "-mcpu=cortex-m4", "-mthumb", "-T", "targets/nrf52.ld", "-Wl,--gc-sections", "-fno-exceptions", "-fno-unwind-tables", "-ffunction-sections", "-fdata-sections", "-Os", "-DNRF52832_XXAA", "-Ilib/CMSIS/CMSIS/Include", "lib/nrfx/mdk/system_nrf52.c", "src/device/nrf/nrf52.s"],
+       "pre-link-args": ["-nostdlib", "-nostartfiles", "-mcpu=cortex-m0", "-mthumb", "-T", "targets/nrf52.ld", "-Wl,--gc-sections", "-fno-exceptions", "-fno-unwind-tables", "-ffunction-sections", "-fdata-sections", "-Os", "-DNRF52832_XXAA", "-Ilib/CMSIS/CMSIS/Include", "lib/nrfx/mdk/system_nrf52.c", "src/device/nrf/nrf51.s"],
        "objcopy": "arm-none-eabi-objcopy",
        "flash": "nrfjprog -f nrf52 --sectorerase --program {hex} --reset"
 }
```

deadprogram, 2018-09-30:
Error received in compile:

```
$ make flash-serial TARGET=microbit
./build/tgo build -target microbit -size=short -o build/serial.elf examples/serial
/tmp/tinygo517145424/main.o: In function `runtime.ticks':
/home/ron/.gvm/pkgsets/go1.11/global/src/github.com/aykevl/tinygo/src/runtime/runtime_nrf.go:78: undefined reference to `__umodsi3'
collect2: error: ld returned 1 exit status
error: exit status 1
Makefile:97: recipe for target 'build/serial.elf' failed
make: *** [build/serial.elf] Error 1
```

deadprogram, 2018-09-30:
I forgot to remove `-nostdlib` from `pre-link-args`.

Now back to this compile error:
```
$ make flash-serial TARGET=microbit
go build -o build/tgo -i .
./build/tgo build -target microbit -size=short -o build/serial.elf examples/serial
/usr/lib/gcc/arm-none-eabi/4.9.3/armv6-m/libgcc.a(unwind-arm.o): In function `get_eit_entry':
/build/gcc-arm-none-eabi-DRRnLA/gcc-arm-none-eabi-4.9.3+svn231177/build/arm-none-eabi/armv6-m/libgcc/../../../../src/libgcc/unwind-arm-common.inc:274: undefined reference to `__exidx_start'
/build/gcc-arm-none-eabi-DRRnLA/gcc-arm-none-eabi-4.9.3+svn231177/build/arm-none-eabi/armv6-m/libgcc/../../../../src/libgcc/unwind-arm-common.inc:274: undefined reference to `__exidx_end'
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/armv6-m/libc.a(lib_a-abort.o): In function `abort':
/build/newlib-5zwpxE/newlib-2.2.0+git20150830.5a3d536/build/arm-none-eabi/armv6-m/newlib/libc/stdlib/../../../../../../newlib/libc/stdlib/abort.c:63: undefined reference to `_exit'
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/armv6-m/libc.a(lib_a-signalr.o): In function `_kill_r':
/build/newlib-5zwpxE/newlib-2.2.0+git20150830.5a3d536/build/arm-none-eabi/armv6-m/newlib/libc/reent/../../../../../../newlib/libc/reent/signalr.c:61: undefined reference to `_kill'
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/armv6-m/libc.a(lib_a-signalr.o): In function `_getpid_r':
/build/newlib-5zwpxE/newlib-2.2.0+git20150830.5a3d536/build/arm-none-eabi/armv6-m/newlib/libc/reent/../../../../../../newlib/libc/reent/signalr.c:97: undefined reference to `_getpid'
collect2: error: ld returned 1 exit status
error: exit status 1
Makefile:97: recipe for target 'build/serial.elf' failed
make: *** [build/serial.elf] Error 1
```

aykevl, 2018-09-30:
Removing `-nostdlib` should not be necessary anymore with master, the `__umodsi3` reference is only the side effect of a now-fixed bug.

deadprogram, 2018-09-30:
You are correct, it builds correctly on master. Still nothing seems to be happening.

aykevl, 2018-10-02:
My micro:bit arrived :D

For me the same, nothing happens. Will have to investigate what goes wrong, probably with OpenOCD + GDB.

aykevl, 2018-10-03:
I have improved the debugging experience a lot with the last few commits ([diff](https://github.com/aykevl/tinygo/compare/ef2ac095611dd857327074e1e83eaf9d9d3bcc22...9d408b7cbce6558be21a7b06ea98978f7d18a932)). With that, I was able to pinpoint where the micro:bit gets stuck:

```
$ tinygo gdb -target microbit examples/serial
GNU gdb (7.12-6+9+b2) 7.12.0.20161007-git
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-linux-gnu --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /tmp/tinygo292972977/main...done.
Remote debugging using :3333
runtime.initLFCLK () at /home/ayke/src/github.com/aykevl/tinygo/src/runtime/runtime_nrf.go:40
40		for nrf.CLOCK.EVENTS_LFCLKSTARTED == 0 {
Loading section .text, size 0x35c lma 0x0
Start address 0x0, load size 860
Transfer rate: 1 KB/sec, 860 bytes/write.
Continuing.
WARNING! The target is already running. All changes GDB did to registers will be discarded! Waiting for target to halt.
^C
Program received signal SIGINT, Interrupt.
0x000001de in runtime.initLFCLK () at /home/ayke/src/github.com/aykevl/tinygo/src/runtime/runtime_nrf.go:40
40		for nrf.CLOCK.EVENTS_LFCLKSTARTED == 0 {
(gdb) bt
#0  0x000001de in runtime.initLFCLK () at /home/ayke/src/github.com/aykevl/tinygo/src/runtime/runtime_nrf.go:40
#1  runtime.init#1 () at /home/ayke/src/github.com/aykevl/tinygo/src/runtime/runtime_nrf.go:25
#2  runtime.initAll () at /home/ayke/src/github.com/aykevl/tinygo/src/runtime/runtime.go:11
#3  0x00000218 in main () at /home/ayke/src/github.com/aykevl/tinygo/src/runtime/runtime.go:33
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) list
35	}
36	
37	func initLFCLK() {
38		nrf.CLOCK.LFCLKSRC = nrf.CLOCK_LFCLKSTAT_SRC_Xtal
39		nrf.CLOCK.TASKS_LFCLKSTART = 1
40		for nrf.CLOCK.EVENTS_LFCLKSTARTED == 0 {
41		}
42		nrf.CLOCK.EVENTS_LFCLKSTARTED = 0
43	}
44	
(gdb) quit
Detaching from program: /tmp/tinygo292972977/main, Remote target
Ending remote debugging.
```

When I commented out the `initLFCLK()` and fixed the UART pins, I was able to get a 'hello world!' on my serial monitor :grinning: Of course, sleep doesn't work yet so that should still be fixed.

deadprogram, 2018-10-03:
OK that is very exciting!

aykevl, 2018-10-05:
After some digging in old MicroPython source code, I finally figured out the problem with the clock: the micro:bit does not have a 32kHz crystal. I haven't tested it yet but I'm pretty sure that's it. The solution is simply to use the internal RC oscillator, which uses slightly more power and is not very precise without calibration.

deadprogram, 2018-10-05:
Just wondering if `-DNRF52832_XXAA` should be `-DNRF51` in your `microbit.json`?