tinygo-org/tinygo #35: tinygo.org

aykevl, 2018-10-03:
I just discovered that someone has registered tinygo.org and redirected it to this repository.

I'm going to assume that was only meant to be helpful, however I'd really like to keep such things centralized. And at least I would have liked to know that this domain was registered and who owns it now.
So here is a request: can whoever registered the domain contact me? Either here in the issue tracker or by mail. I'm reachable at aykevanlaethem@gmail.com. Then we can discuss what to do with it, for example host a real homepage.

deadprogram, 2018-10-04:
That was me, email already sent. 

Just trying to be helpful to the long-term project, by making it easier for people to find and contribute to.

aykevl, 2018-10-04:
Okay then it's fine :) I was just worried some random person registered it without real connections to the project.

closed 2018-10-04 by aykevl