tinygo-org/tinygo #46: What does "no garbage collection" mean?

Xe, 2018-10-19:
From the readme: 

> not yet supported:
> ...
> - garbage collection

What does this mean? Does this mean that all values are on stack memory? Does this mean that all pointers are allocated on the heap and never freed?

aykevl, 2018-10-19:
The latter. All pointers are allocated and never freed. However, there is a simple heap-to-stack pass that tries to allocate objects on the stack if possible so that the most obvious cases get stack-allocated.

aykevl, 2018-10-19:
Closing as this is a question, not an issue. You are of course free to open an issue for GC support.

closed 2018-10-19 by aykevl

kyegupov, 2019-01-09:
A note re wasm target: since webassembly has no stack introspection, marksweep cannot be ported to it. Three possible (all imperfect strategies are: manual free (incl. optionally delegating alloc/free to host), reference counting, semi-manual allocation of objects into "permanent" and "temporary" heaps.

aykevl, 2019-01-09:
I think marksweep can be ported with some compiler work. LLVM has some support for it. It just needs to be done, and it isn't easy. See https://llvm.org/docs/GarbageCollection.html, the most likely candidate would be the shadow stack GC, at least initially.

kyegupov, 2019-01-14:
@aykevl is there any timeline/roadmap for implementing garbage collection?

aykevl, 2019-01-14:
On WebAssembly I assume?
Not really, certainly after #130 (almost finished), maybe after #104 (partially done). It's also hard to say how long the feature itself is going to take as I'm not sure how difficult it is.

wilsonwang371, 2022-05-25:
does tinygo support GC in WASM now? Do I have to call GC in my code to get memory garbage collected?

aykevl, 2022-05-25:
Yes, TinyGo supports GC in WebAssembly. And you don't need to call `runtime.GC` for it, it happens automatically.