tinygo-org/tinygo #47: OS X

aykevl, 2018-10-20:
Currently, TinyGo doesn't appear to work on OS X with LLVM 7 from homebrew.

  * Support OS X
  * Optional: build on OS X in Travis

aykevl, 2018-10-22:
See https://github.com/aykevl/go-llvm/pull/1

aykevl, 2018-10-27:
I think tinygo works now on OS X after installing LLVM with homebrew - but it doesn't support WebAssembly there yet.

deadprogram, 2018-11-05:
Can this issue now be closed? Anyone able to confirm using Mojave? I am running High Sierra on my test Mac at present.

deadprogram, 2018-11-05:
Oh, seems like the WASM support is not in there yet?

aykevl, 2018-11-05:
That's right, homebrew does not include WebAssembly by default (they think it's unstable, see https://github.com/Homebrew/homebrew-core/pull/29068). Also, I assume they haven't enabled the AVR backend either. I don't know much about OS X or homebrew so you should probably look elsewhere for enabling these backends.

But based on your comment it seems to be basically working? In that case I'm closing the issue.

closed 2018-11-05 by aykevl