tinygo-org/tinygo #50: Support MCU development with Docker

aykevl, 2018-10-21:
Currently, the generated Docker image misses a few things important for microcontroller development:

  * autogenerated register descriptors (`make gen-device`)
  * build tools, like binutils, gcc, etc.

Preferably, ARM targets should move to Clang first so we don't depend on GCC, avoiding the extra dependency.

deadprogram, 2018-10-25:
Probably would be good to have a multi-stage build so the MCU docker and the WASM docker files do not have to bring down all of the unique dependencies of the other.

Also, we would probably need to run the MCU docker with the `--device` flag so we can flash devices, whereas that is not needed for other use cases.

deadprogram, 2018-11-08:
Just submitted PR https://github.com/aykevl/tinygo/pull/72 to address this issue.

Note that this PR does not attempt to install tools for flashing. Instead, you would use the image for compiling, and then flash the resulting ELF file from your native OS.

aykevl, 2018-11-15:
Fixed with #72.

closed 2018-11-15 by aykevl