tinygo-org/tinygo #60: How to call CGo from TinyGo

deadprogram, 2018-10-25:
I would like to be able to call CGo from TinyGo, for example the NRF5 sdk.

Is there a way I can currently do this from TinyGo?

aykevl, 2018-10-25:
There is very limited support for CGo. You can call functions, but I don't think you can do much more than that. But note that CGo cannot be used in the runtime or in the machine package,due to import cycles.

If it doesn't work, what is the error you're getting?

deadprogram, 2018-11-15:
So I tried to compile a very simple CGo program and received this error:

```
$ tinygo build -target=microbit -o=cxp.hex ./src/examples/cxp/cxp.go 
/home/ron/.gvm/pkgsets/go1.11/global/src/github.com/aykevl/tinygo/src/examples/cxp/cxp.go:8:8: could not import C (the loader doesn't cgo-process ad hoc packages like "main"; see Go issue 11627)
error: couldn't load packages due to errors: main
```

My code is:

```go
//#cgo CFLAGS: -I/home/ron/.gvm/pkgsets/go1.11/global/src/github.com/aykevl/tinygo/src/examples/cxp

package main

/*
#include "hello.h"
*/
import "C"

func main() {
	println(C.fortytwo())
}
```

with the following C code:

hello.h:
```c
int fortytwo();
```

and hello.c
```c
#include "hello.h"

int fortytwo()
{
	return 42;
}
```

I added the line `"src/examples/cxp/hello.c"` to the end of the `"pre-link-args"` section of the `microbit.json` file as well.

aykevl, 2018-11-15:
>     could not import C (the loader doesn't cgo-process ad hoc packages like "main"; see Go issue 11627)

This is probably the error. Ad-hoc packages are packages created when passing a .go file directly on the command line instead of a fully-qualified Go package path. Try replacing "./src/examples/cxp/cxp.go" with "examples/cxp" in the command line.

deadprogram, 2018-11-15:
OK new error:

```
$ tinygo build -target=microbit -o=cxp.hex examples/cxp 
function declaration may not have a !dbg attachment
i32 ()* @fortytwo
error: function declaration may not have a !dbg attachment
i32 ()* @fortytwo
```

aykevl, 2018-11-15:
... interesting. A quick workaround may be adding `-no-debug` to the command line. But I'm curious what causes the error here. Can you share how you declare/define `fortytwo`?

deadprogram, 2018-11-15:
See https://github.com/aykevl/tinygo/issues/60#issuecomment-439144665

deadprogram, 2018-11-15:
Running `tinygo build -target=microbit -o=cxp.hex -no-debug examples/cxp` compiled without error. Now I need to make the code to do something visible, to verify it is working.

deadprogram, 2018-11-15:
OK this works!

```go
//#cgo CFLAGS: -I/home/ron/.gvm/pkgsets/go1.11/global/src/github.com/aykevl/tinygo/src/examples/cxp

package main

/*
#include "hello.h"
*/
import "C"

func main() {
	for {
		println(C.fortytwo())
	}
}
```

deadprogram, 2018-11-15:
Note that the `-no-debug` was required to compile that code.

aykevl, 2018-11-15:
A fix is being tested by Travis here: https://github.com/aykevl/tinygo/commit/1ed019771d4b44d0def11c6b8ae3e08d33d3cca2

aykevl, 2018-11-15:
Glad to hear it works!
It would be nice to make this easier in the future. But, you know, so many things to do.

deadprogram, 2018-11-15:
Had to get it working first, of course. Now I can try to use it for something slightly more real.

aykevl, 2018-12-01:
See #89.

deadprogram, 2019-01-11:
We should create a page on the web site to collect the current info on using CGo with TinyGo, so that this issue can be closed, and specific individual ones opened as needed.

deadprogram, 2019-02-08:
New capabilities fro CGo have just been merged into the `dev` branch.

aykevl, 2019-05-30:
CGo support was again improved a lot in the latest release. It is now starting to be usable.

torntrousers, 2020-08-11:
Is there any doc or other information on the state of using C other than this issue?

aykevl, 2020-08-11:
Not really. I would suggest trying things out and seeing whether it works or not.
Known limitations:

- No `C.GoString` and related functions.
- Limited support for C macros (for example, `#define FOO (3 + 4)` won't work, but `#define FOO (3)` would).

aykevl, 2020-08-11:
You can also take a look at the TinyGo Bluetooth package, which relies heavily on CGo for SoftDevice support (nrf chips): https://github.com/tinygo-org/bluetooth

torntrousers, 2020-08-12:
Ok thanks, I might give it a try. Do you think it could be possible to use this to call existing sensor libraries written in C instead of having to rewrite them in TinyGo?

aykevl, 2020-08-13:
Maybe. The best way to know that is to try it out.

dmlambea, 2021-06-24:
Hi, @deadprogram 

This thread seems to be the only source of information regarding the linking of C code with TinyGo code, and I'm still missing parts of the recipe. Maybe you know where should I look for information on this topic.

My case is quite similar to yours:  I want to call some C code from main.go and I can compile the program but during the linkage some errors arise saying that there are missing symbols. I've tried to add the .c files to a `pre-link-args` in the JSON target with no success. I've also compiled the C code by hand using the `arm-none-eabi-gcc` compiler and using the .o files in the `pre-link-args` but I see no difference, like if the array didn't existed.

The error is:
```
tinygo:ld.lld: error: undefined symbol: TestFunc
>>> referenced by /tmp/tinygo253258736/main.o:(runtime.run$1$gowrapper)
failed to run tool: ld.lld
error: failed to link /tmp/tinygo253258736/main: exit status 1
```

I'm writing to you because it seems that you succeeded on this. Your help is greatly appreciated!

deadprogram, 2021-06-24:
I suggest following @aykevl advice and look at the Bluetooth package https://github.com/tinygo-org/bluetooth

Also check out https://github.com/tinygo-org/tinygo/tree/release/cgo/testdata for some examples that are part of our automated tests.

Hope that helps!

aykevl, 2021-06-25:
@dmlambea can you share the source code that shows this error? I cannot say what is going wrong from the error message.

Note that TinyGo does not currently support functions in the header file. So something like this does not work:

```go
// int TestFunc(int a, int b) {
//     return a + b;
// }
import "C"
```

Instead you need to move the C code in a separate file in the same package.

bgould, 2021-06-25:
https://github.com/tinygo-org/tinyfs also demonstrates CGo usage from TinyGo, both the littlefs and fatfs packages are wrappers around C libraries

dmlambea, 2021-06-27:
Thanks, @deadprogram @aykevl @bgould !

I had reviewed those sources (the Bluetooth and the testdata) before but I was not able to extract my case from there. The bluetooth package calls the SoftDevice by using SVCALL in ASM and the testdata seems to contain just headers/definitions, but no "real" C code  (bear with me here, I mean "real" C code to something more than defining types and/or constants).

On the other hand, the tinyfs project and @aykevl 's suggestion of moving the C code to the same package of the Go files were really useful. Simple C code is now compiling along with the TinyGo source code with no extra efforts, thanks!

Just one more question: is it possible to link already-compiled code to a TinyGo program? Not always the C code will be available in source form (or ready to be cleanly moved to TinyGo's source tree).

dmlambea, 2021-07-14:
I guess that currently there is not an easy way of linking a tinygo program against a .so lib. Are there any plans of making this feature in a near future?

Summarizing, I need to know how to link vendor's precompiled code (i.e., no access to C source code) agains a program in TinyGo. I've seen the linker scripts in TinyGo installation folder but manually dealing with it seems a bit too lowlevel for TinyGo's regular abstraction level. Might it be the only way to go nowadays?

deadprogram, 2021-11-06:
Closing, since this has been addressed for a while. Please open new specific issues as needed. Thanks!

closed 2021-11-06 by deadprogram

aykevl, 2021-11-10:
> I guess that currently there is not an easy way of linking a tinygo program against a .so lib. Are there any plans of making this feature in a near future?

What you can do is use `#cgo LDFLAGS:` lines to add some linker flags to link with the given .so file. I haven't tried this specifically but I think it's possible that way.
You certainly don't need to use linker scripts for this purpose: linker scripts are not meant for this purpose. They are used to define custom memory layouts (especially on baremetal systems).

Bmixo, 2022-12-31:
It's almost 2023, how is going to support generate .dll or l.ib