tinygo-org/tinygo #64: wasm: slice copy bombing out

trashhalo, 2018-10-29:
A little toy program I was working on keeps hitting an error in the webassembly code for slicecopy. I don't see anything obviously wrong, so it might be a bug. happens if I try and call `updatePage` from javascript.

```go
package main

import (
	"strconv"
	"strings"
)

type Message int

const (
	Draw Message = iota
)

func main() {
}

//go:export updatePage
func updatePage(x, y int) string {
	page := strings.Join([]string{
		"<div style=\"",
		"position:absolute; ",
		"width:50px; ",
		"height:50px; ",
		"background: red; ",
		"left: ",
		strconv.Itoa(x),
		"px; top: ",
		strconv.Itoa(y),
		"px;\"></div>",
	}, "")
	return page
}
```

aykevl, 2018-10-29:
Yes, that looks like a bug. Currently investigating.

However, note that you can't currently return a string from WebAssembly. This is a limitation of WebAssembly at the moment. In my experiments, it doesn't return anything at the moment, probably because it stores a pointer to the result somewhere but I don't know where exactly.

closed 2018-10-29 by aykevl

aykevl, 2018-10-29:
It was an incorrect backwards copy of a zero-length slice: even while being zero length, it tried to copy one byte. The bug should be fixed now.

I still don't understand why the first parameter from JS is skipped. It looks like it happens during WebAssembly codegen (after all optimization passes etc. have run) because the function signature in the IR is correct: two i32 parameters.