tinygo-org/tinygo #65: wasm: panic: runtime error: slice out of range

trashhalo, 2018-10-30:
panic: runtime error: slice out of range

runtime.runtimePanic (wasm-77de193e-12:8)
runtime.sliceBoundsCheck (wasm-77de193e-13:15)
(*bytes.Buffer).tryGrowByReslice (wasm-77de193e-17:37)
(*bytes.Buffer).WriteString (wasm-77de193e-16:18)
mouse (wasm-77de193e-11:10)
document.getElementById.onmousemove (wasm.js:61)

===
https://github.com/trashhalo/tiny-wasm/blob/master/wasm.go
You can reproduce this issue by cloning the repo above
1. `make deps`
2. `make build`
3. `./http`
4. Send your browser to http://localhost:8080 and move your mouse around the red box and see panics in console

closed 2018-10-30 by aykevl

aykevl, 2018-10-30:
Fixed. The compiler was inserting a runtime slice bounds check based on the length of the slice instead of the capacity - as it should have been according to the spec. Apparently `*bytes.Buffer` reslices an internal buffer outside of the length but inside the capacity, surfacing this bug.

I've also made a few other changes to the compiler that should make it easier to detect this error (by executing a WebAssembly `unreachable` instruction on such errors) and I made sure the custom Docker image with a custom wasm.syms file should not be necessary anymore. You can see the build status here: https://hub.docker.com/r/tinygo/tinygo/builds/

aykevl, 2018-10-30:
Thanks again for the report, these are the kind of bugs that need to be fixed to make TinyGo stable.