tinygo-org/tinygo #66: wasm: importing image package causes unknown convert

trashhalo, 2018-10-31:
```
        errors.New: return t2
        bufio.init: *ErrAdvanceTooFar = t8
        bufio.init: t9 = errors.New("final token":string)
        errors.New: t0 = new errorString (complit)
        errors.New: t1 = &t0.s [#0]
        errors.New: *t1 = text
        errors.New: t2 = make error <- *errorString (t0)
        errors.New: return t2
        bufio.init: *ErrFinalToken = t9
        bufio.init: t10 = convert []byte <- string ("�":string)
error: todo: init: unknown convert: convert []byte <- string ("�":string)
Makefile:9: recipe for target 'build' failed
make: *** [build] Error 1
```
Reproduction repo
https://github.com/trashhalo/tiny-wasm/blob/images/wasm.go
1. make deps
2. make build

aykevl, 2018-10-31:
Looks like this line is causing the error:
https://golang.org/src/bufio/scan.go#L291

aykevl, 2018-10-31:
It is likely fixed by 1b283c11c11a632618943403069e49c9480d9f17, but I haven't tested it.

trashhalo, 2018-11-01:
I can confirm this error no longer happens on master.

closed 2018-11-03 by aykevl