tinygo-org/tinygo #67: wasm: encoding/png causes reflect errors during import

trashhalo, 2018-10-31:
```
➜  tiny-wasm git:(encoding-png) ✗ make build
docker run --rm -v /home/stephen/go/src/github.com/trashhalo/tiny-wasm:/go/src/github.com/trashhalo/tiny-wasm tinygo \
build -o /go/src/github.com/trashhalo/tiny-wasm/wasm.wasm -target wasm -dumpssa github.com/trashhalo/tiny-wasm
/usr/local/go/src/encoding/binary/binary.go:587:26: invalid operation: v (variable of type reflect.Value) has no field or method CanSet
/usr/local/go/src/encoding/binary/binary.go:522:26: invalid operation: v (variable of type reflect.Value) has no field or method CanSet
/usr/local/go/src/encoding/binary/binary.go:397:17: invalid operation: t (variable of type reflect.Type) has no field or method Len
/usr/local/go/src/encoding/binary/binary.go:402:20: invalid operation: t (variable of type reflect.Type) has no field or method NumField
/usr/local/go/src/encoding/binary/binary.go:403:27: invalid operation: t.Field(i) (value of type reflect.StructField) has no field or method Type
/usr/local/go/src/encoding/binary/binary.go:415:16: invalid operation: t (variable of type reflect.Type) has no field or method Size
/usr/local/go/src/encoding/binary/binary.go:375:26: Indirect not declared by package reflect
/usr/local/go/src/encoding/binary/binary.go:359:15: Indirect not declared by package reflect
/go/src/github.com/trashhalo/tiny-wasm/wasm.go:22:23: too many arguments
error: couldn't load packages due to errors: github.com/trashhalo/tiny-wasm, encoding/binary
Makefile:9: recipe for target 'build' failed
make: *** [build] Error 1
```
https://github.com/trashhalo/tiny-wasm/blob/encoding-png/wasm.go

aykevl, 2018-10-31:
It appears that the image/png package does not actually require the reflect package - it only needs some functions from encoding/binary that don't use reflection.

If you want to fix this compiler error, you can try stubbing out the required functions: https://github.com/aykevl/tinygo/tree/master/src/reflect

aykevl, 2018-11-03:
I merged the changes you made to the reflect package in https://github.com/aykevl/tinygo/commit/22914165ccff0f7ea72dd5f7b07d5f46dc72bd6e. Thank you for that!

It looks like more work is needed before encoding/png can compile, in particular #62 would make this a whole lot easier (it probably fixes the last error you reported). While testing with that branch, I noticed that `(*image/png.decoder).decode` has a type of `defer` statement that is not yet supported: it defers an interface call. Should not be a whole lot of work but it indicates that encoding/png is still a bit of work to get running.

aykevl, 2018-11-03:
Defer of interfaces is now implemented: https://github.com/aykevl/tinygo/commit/a531caa2e9f94a482da49b890b0089299a31b3fc

Next up is `[]byte(string)` in #62 (compile with `-initinterp` in that branch).

aykevl, 2018-11-04:
Found and fixed the bug, but fixing it turned up even more issues that were hidden before.

I think the conclusion is that image/png will take a while before it is supported. It depends on lots of packages (compress/flate in particular) that do funky stuff at package initialization time. Maybe you can use a different image format until image/png is supported? I haven't tried it, but you could try golang.org/x/image/bmp.

deadprogram, 2019-01-11:
Does this issue still have a specific purpose, or can it now be closed?