tinygo-org/tinygo #68: The "time" package

j7b, 2018-11-03:
With docker image:

```
cat demo.go && tinygo run demo.go && date +%s
package main

import "time"

func main() {
        println(time.Now().Unix())
}
22598
1541263654
```
Off by orders of magnitude, nanoseconds output also looks suspicious.
```
cat demo2.go && tinygo run demo2.go 
package main

import "time"

func main() {
        println(time.Now().String())
}
Call parameter type does not match function signature!
  %101 = extractvalue { i8, i1 } %100, 0
 i32  call void @runtime.sliceBoundsCheck(i32 %109, i8 %101, i32 %108)
Both operands to a binary operator are not of the same type!
  %111 = sub i32 %108, i8 %101
Both operands to a binary operator are not of the same type!
  %112 = sub i32 %109, i8 %101
LLVM ERROR: Broken module found, compilation aborted!
```

aykevl, 2018-11-03:
Actually, these are two separate bugs.

1. This is because TinyGo uses monotonic time instead of a real timestamp. This should be fixed.
2. Apparently I missed a zero/sign extension somewhere.

aykevl, 2018-11-14:
The LLVM error you see should be fixed in master. However, `time.Now().String()` now panics at runtime because system calls have not yet been implemented.

deadprogram, 2019-09-19:
So, an update on the various issues reported here, as of v0.8.0:

- TinyGo still uses monotonic time instead of a real timestamp on Linux, so that needs to be addressed.
- `time.Now().String()` now returns without error:
```
$ tinygo run demo.go && date +%s
1970-01-04 02:29:33.486739311 +0100 CET m=+264573.486739312
1568879646
```

So I think that the only remaining issue is the monotonic time.

deadprogram, 2020-07-05:
Monotonic time is still an issue when running on Linux:

```
$ tinygo run demo.go
1970-01-02 18:46:00.462465471 +0100 CET m=+150360.462465472
```

aykevl, 2021-07-20:
Here is a fix: https://github.com/tinygo-org/tinygo/pull/2007

deadprogram, 2021-09-21:
Released with 0.20.0 so now closing. Thank you!

PR #2007 (2021-07-20): runtime: fix time base for time.Now()
This function previously returned the atomic time, that isn't affected
by system time changes but also has a time base at some arbitrary time
in the past. This makes sense for baremetal platforms (which typically
don't know the wall time) but it gives surprising results on Linux and
macOS: time.Now() usually returns a time somewhere near the start of
1970.

This commit fixes this by obtaining both time values: the monotonic time
and the wall clock time. This is also how the Go runtime implements the
time.now function.

---

This fixes #68.

Note that with  [... cut]
closed 2021-09-21 by deadprogram