tinygo-org/tinygo #69: String to []rune conversion

j7b, 2018-11-03:
```
cat demo.go && tinygo run demo.go 
package main

func main() {
        runes := []rune("Hello Sailor")
        copy(runes,[]rune("PLUGH"))
}

error: compiler: todo: convert from string: rune
```

aykevl, 2018-12-01:
Not yet fixed, but I've improved the error message to also include the source location (see a00ab0fbc7c75864de946fe64bed7c2b3594082e):

```
$ tinygo build -o wasm.wasm -target=wasm examples/wasm
/home/ayke/src/github.com/aykevl/tinygo/src/examples/wasm/wasm.go:26:12: todo: convert from string: rune
```

scriptonist, 2019-05-29:
Hey, I would like to take a stab at this.

aykevl, 2019-05-29:
That would be great!

The feature will be very similar to the current `stringFromBytes` implementation converting a string to a `[]byte` slice. The difference with converting to a `[]rune` slice is that that needs UTF-8 decoding, but all the decoding stuff has already been implemented in the runtime so you can take a look at what's already there.
These are the places that need some additions:

https://github.com/tinygo-org/tinygo/blob/da857108947aa35ff6e6b5ce16c96359e72fe47c/src/runtime/string.go#L91
https://github.com/tinygo-org/tinygo/blob/da857108947aa35ff6e6b5ce16c96359e72fe47c/compiler/compiler.go#L2387

Just ask if you have any questions.

scriptonist, 2019-05-30:
Thanks @aykevl, Let me give it a try.

aykevl, 2019-06-18:
Fixed by #406.

closed 2019-06-18 by aykevl