tinygo-org/tinygo #78: wasm: importing math/rand has init issues

trashhalo, 2018-11-13:
error: todo: init call: math/rand.NewSource

```
        math/rand.init: *t1748 = -4300543082831323144:int64
        math/rand.init: *t1749 = -6344160503358350167:int64
        math/rand.init: *t1750 = 5896236396443472108:int64
        math/rand.init: *t1751 = -758328221503023383:int64
        math/rand.init: *t1752 = -1894351639983151068:int64
        math/rand.init: *t1753 = -307900319840287220:int64
        math/rand.init: *t1754 = -6278469401177312761:int64
        math/rand.init: *t1755 = -2171292963361310674:int64
        math/rand.init: *t1756 = 8382142935188824023:int64
        math/rand.init: *t1757 = 9103922860780351547:int64
        math/rand.init: *t1758 = 4152330101494654406:int64
        math/rand.init: t1759 = new lockedSource (complit)
        math/rand.init: t1760 = &t1759.src [#1]
        math/rand.init: t1761 = NewSource(1:int64)
Makefile:9: recipe for target 'build' failed
make: *** [build] Error 1
```
```go
// +build wasm,!arm,!avr

package main

import (
	_ "math/rand"
)

func main() {
}
```

aykevl, 2018-11-13:
Have you tried the `-initinterp` flag?
I haven't enabled it by default yet because I'm concerned there are programs that may stop compiling with it, but will probably enable it soon.

aykevl, 2018-11-30:
This specific issue should not exist anymore with the new init system.

closed 2018-11-30 by aykevl