tinygo-org/tinygo #82: WASM Target Not Supported on OSX

drush, 2018-11-15:
Adding a tracking case for monitoring progress of WASM target when building from OSX.  

OSX developers cannot use tinygo to build WASM targets because WASM is an experimental feature in LLVM
```
$ tinygo build -o test.wasm test.go
error: No available targets are compatible with this triple.
```

The Facts:
* Tinygo can build on OSX when LLVM 7 is installed via brew.  However, WASM targets are experimental in LLVM 7 and not supported by the homebrew recipe.
* WASM will be supported in LLVM 8 (Expected release March 2019)

RELATED:
* #47 OSX Support
* [Tracking LLVM 8 progress](https://llvm.org/docs/ReleaseNotes.html)

Possible Workarounds:
* Wait for LLVM 8 in March of 2019
* Build LLVM 7 from source with full WASM support (Does this work?)
* ???

deadprogram, 2018-11-15:
Hi @drush been quite a while, hope all is well.

You can use the Docker image to build WASM on macOS. Check out https://tinygo.readthedocs.io/en/latest/docker.html for more info.

deadprogram, 2019-01-07:
We now have a Homebrew brew for installing LLVM with the required dependencies to run on macOS: https://github.com/tinygo-org/homebrew-tools

Assuming that this works fine for everyone, we can update the docs site accordingly.

peterhellberg, 2019-01-07:
Unfortunately `brew reinstall tinygo-org/tools/llvm` errored out with this output under macOS High Sierra (10.13.6), is Mojave required?

```sh
==> cmake -G Unix Makefiles .. -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FL
==> make

==> make install
==> make install-xcode-toolchain
Last 15 lines from /Users/peter/Library/Logs/Homebrew/llvm/04.make:
gmake[9]: Leaving directory '/private/tmp/llvm-20190107-34453-j3axuk/llvm-7.0.1.src/build/tools/clang/runtime/compiler-rt-bins'
[  7%] Built target RTXrayFDR.osx
gmake[8]: Leaving directory '/private/tmp/llvm-20190107-34453-j3axuk/llvm-7.0.1.src/build/tools/clang/runtime/compiler-rt-bins'
gmake[7]: *** [Makefile:130: all] Error 2
gmake[7]: Leaving directory '/private/tmp/llvm-20190107-34453-j3axuk/llvm-7.0.1.src/build/tools/clang/runtime/compiler-rt-bins'
gmake[6]: *** [tools/clang/runtime/CMakeFiles/compiler-rt.dir/build.make:116: tools/clang/runtime/compiler-rt-stamps/compiler-rt-build] Error 2
gmake[6]: Leaving directory '/private/tmp/llvm-20190107-34453-j3axuk/llvm-7.0.1.src/build'
gmake[5]: *** [CMakeFiles/Makefile2:35934: tools/clang/runtime/CMakeFiles/compiler-rt.dir/all] Error 2
gmake[5]: Leaving directory '/private/tmp/llvm-20190107-34453-j3axuk/llvm-7.0.1.src/build'
gmake[4]: *** [Makefile:155: all] Error 2
gmake[4]: Leaving directory '/private/tmp/llvm-20190107-34453-j3axuk/llvm-7.0.1.src/build'
make[3]: *** [tools/xcode-toolchain/CMakeFiles/build-xcode-toolchain] Error 2
make[2]: *** [tools/xcode-toolchain/CMakeFiles/build-xcode-toolchain.dir/all] Error 2
make[1]: *** [tools/xcode-toolchain/CMakeFiles/install-xcode-toolchain.dir/rule] Error 2
make: *** [install-xcode-toolchain] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
https://github.com/tinygo-org/homebrew-tools/issues

Error: A newer Command Line Tools release is available.
Update them from Software Update in the App Store.
```

deadprogram, 2019-01-07:
Hi @peterhellberg did you try running `xcode-select --install` to update your XCode CLI tools?

peterhellberg, 2019-01-07:
@deadprogram I upgraded the command line tools via Xcode (10.1) and ran `brew install tinygo-org/tools/llvm` again. (Got the same error output)

```sh
$ brew install tinygo-org/tools/llvm
Updating Homebrew...
Fast-forwarded master to origin/master.
Fast-forwarded master to origin/master.
Fast-forwarded master to origin/master.
Fast-forwarded master to origin/master.
==> Auto-updated Homebrew!
Updated Homebrew from 6cd4c317 to 464b41eb.
Updated 3 taps (homebrew/cask-versions, homebrew/core and homebrew/cask).
==> Updated Formulae
mkcert ✔               eslint                 fonttools              libsodium

==> Installing llvm from tinygo-org/tools
==> Downloading https://releases.llvm.org/7.0.1/llvm-7.0.1.src.tar.xz
Already downloaded: /Users/peter/Library/Caches/Homebrew/downloads/deef0de57101b23ba53aba53b9b7a08ea1b48cfd5e538dd34119e24f9d8ed6e0--llvm-7.0.1.src.tar.xz
==> Downloading https://releases.llvm.org/7.0.1/cfe-7.0.1.src.tar.xz
Already downloaded: /Users/peter/Library/Caches/Homebrew/downloads/e88421c4653fa9f4ffce3393a1f6d5142d07908c95d1f75a773666512dbe373b--cfe-7.0.1.src.tar.xz
==> Downloading https://releases.llvm.org/7.0.1/clang-tools-extra-7.0.1.src.tar.xz
Already downloaded: /Users/peter/Library/Caches/Homebrew/downloads/c055cc8e0627d814063653c11d38b48b74bc35a0b0d39dbf7c65c1972b9c22ee--clang-tools-extra-7.0.1.src.tar.xz
==> Downloading https://releases.llvm.org/7.0.1/openmp-7.0.1.src.tar.xz
Already downloaded: /Users/peter/Library/Caches/Homebrew/downloads/430346c8a0cf1ec6fc94c92b21b0fc3aa2aa857e5731e4208fd8e72859db0222--openmp-7.0.1.src.tar.xz
==> Downloading https://releases.llvm.org/7.0.1/libcxx-7.0.1.src.tar.xz
Already downloaded: /Users/peter/Library/Caches/Homebrew/downloads/0d33e60b8ed8825cb5a22aab98781fafbe296a8bbb8d96df536e794d170cb6e4--libcxx-7.0.1.src.tar.xz
==> Downloading https://releases.llvm.org/7.0.1/libunwind-7.0.1.src.tar.xz
Already downloaded: /Users/peter/Library/Caches/Homebrew/downloads/a1fbaf33a6a9d1b1707d9dfd204a320e91e4196f5e0e137986b83dfb06fbc8d0--libunwind-7.0.1.src.tar.xz
==> Downloading https://releases.llvm.org/7.0.1/lld-7.0.1.src.tar.xz
Already downloaded: /Users/peter/Library/Caches/Homebrew/downloads/28d82806d84a7bbd190fe047904a84918fb8a67efb334b465d683f658a0546f4--lld-7.0.1.src.tar.xz
==> Downloading https://releases.llvm.org/7.0.1/polly-7.0.1.src.tar.xz
Already downloaded: /Users/peter/Library/Caches/Homebrew/downloads/537fd9fbe4f65f37591471e0ad84cb8c02bcbb00b9758b4395bec8facbdfeb3c--polly-7.0.1.src.tar.xz
==> Downloading https://releases.llvm.org/7.0.1/compiler-rt-7.0.1.src.tar.xz
Already downloaded: /Users/peter/Library/Caches/Homebrew/downloads/a82ca2001daf97c74990a94300ccefa01e18013f3a7d33e45b37eb029aaec817--compiler-rt-7.0.1.src.tar.xz
==> cmake -G Unix Makefiles .. -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-D
==> make
==> make install
==> make install-xcode-toolchain
Last 15 lines from /Users/peter/Library/Logs/Homebrew/llvm/04.make:
gmake[9]: Leaving directory '/private/tmp/llvm-20190107-57109-hj7rou/llvm-7.0.1.src/build/tools/clang/runtime/compiler-rt-bins'
[  8%] Built target RTXrayFDR.osx
gmake[8]: Leaving directory '/private/tmp/llvm-20190107-57109-hj7rou/llvm-7.0.1.src/build/tools/clang/runtime/compiler-rt-bins'
gmake[7]: *** [Makefile:130: all] Error 2
gmake[7]: Leaving directory '/private/tmp/llvm-20190107-57109-hj7rou/llvm-7.0.1.src/build/tools/clang/runtime/compiler-rt-bins'
gmake[6]: *** [tools/clang/runtime/CMakeFiles/compiler-rt.dir/build.make:116: tools/clang/runtime/compiler-rt-stamps/compiler-rt-build] Error 2
gmake[6]: Leaving directory '/private/tmp/llvm-20190107-57109-hj7rou/llvm-7.0.1.src/build'
gmake[5]: *** [CMakeFiles/Makefile2:35934: tools/clang/runtime/CMakeFiles/compiler-rt.dir/all] Error 2
gmake[5]: Leaving directory '/private/tmp/llvm-20190107-57109-hj7rou/llvm-7.0.1.src/build'
gmake[4]: *** [Makefile:155: all] Error 2
gmake[4]: Leaving directory '/private/tmp/llvm-20190107-57109-hj7rou/llvm-7.0.1.src/build'
make[3]: *** [tools/xcode-toolchain/CMakeFiles/build-xcode-toolchain] Error 2
make[2]: *** [tools/xcode-toolchain/CMakeFiles/build-xcode-toolchain.dir/all] Error 2
make[1]: *** [tools/xcode-toolchain/CMakeFiles/install-xcode-toolchain.dir/rule] Error 2
make: *** [install-xcode-toolchain] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
https://github.com/tinygo-org/homebrew-tools/issues

Error: A newer Command Line Tools release is available.
Update them from Software Update in the App Store.
```

Another day I'll check if this formula can be installed on a machine running Mojave.

anarcher, 2019-01-08:
> Error: A newer Command Line Tools release is available.
> Update them from Software Update in the App Store.

How about `sudo xcode-select --reset`? 

https://github.com/Homebrew/legacy-homebrew/issues/24014

peterhellberg, 2019-01-10:
@anarcher Did unfortunately not solve the issue (or do anything really since I have the latest version of the CLI-tools installed)

Zambiorix, 2019-02-23:
It works for me (High Sierra 10.13.6) but I get following warnings:

```
VZMS-MBP-2013-Sierra:~ Gerd$ go get -u github.com/tinygo-org/tinygo
# github.com/tinygo-org/tinygo/loader
ld: warning: directory not found for option '-L/usr/lib/llvm-7/lib'
# tinygo.org/x/go-llvm
ld: warning: directory not found for option '-L/usr/local/opt/llvm/lib'
# github.com/tinygo-org/tinygo
ld: warning: directory not found for option '-L/usr/lib/llvm-7/lib'
ld: warning: directory not found for option '-L/usr/local/opt/llvm/lib'
VZMS-MBP-2013-Sierra:~ Gerd$ 
```

And installing llvm7 via brew (https://github.com/tinygo-org/homebrew-tools) gives me the following output (install completes fine)

```
==> llvm7
To use the bundled libc++ please add the following LDFLAGS:
  LDFLAGS="-L/usr/local/opt/llvm7/lib -Wl,-rpath,/usr/local/opt/llvm7/lib"
```

When trying to compile I get the following error :

```
VZMS-MBP-2013-Sierra:tinygo-wasm Gerd$ tinygo build -o wasm.wasm wasm.go
error: failed to link /var/folders/zh/mc43bk2n5nb0cbn8_6y6gr640000gp/T/tinygo536710333/main: exec: "wasm-ld-7": executable file not found in $PATH
```

Not sure what to do next...

Zambiorix, 2019-02-24:
https://github.com/tinygo-org/homebrew-tools

Above link says:

```
brew tap tinygo-org/tools
brew install llvm7
```

shouldn't that be this?

```
brew tap tinygo-org/homebrew-tools
brew install llvm7
```

I uninstalled removed tinygo and uninstalled llvm7 and then:

```
brew tap tinygo-org/tools
brew tap tinygo-org/homebrew-tools
brew install llvm7
```

But then I get the following:

```
==> Tapping tinygo-org/tools
Cloning into '/usr/local/Homebrew/Library/Taps/tinygo-org/homebrew-tools'...
remote: Enumerating objects: 4, done.
remote: Counting objects: 100% (4/4), done.
remote: Compressing objects: 100% (4/4), done.
remote: Total 4 (delta 0), reused 2 (delta 0), pack-reused 0
Unpacking objects: 100% (4/4), done.
Tapped 1 formula (29 files, 38.3KB).
VZMS-MBP-2013-Sierra:~ Gerd$ brew install llvm7
==> Installing llvm7 from tinygo-org/tools
```

I notice that brew is cloning into the correct folder , but it still says
```
==> Tapping tinygo-org/tools
```

not sure where that comes from.
I'll report back when it's done and works or not

deadprogram, 2019-02-24:
Hi @Zambiorix that is part of the default naming for external taps for Homebrew. See https://github.com/Homebrew/brew/blob/master/docs/Taps.md#repository-naming-conventions-and-assumptions for more information.

Zambiorix, 2019-02-24:
thanks @deadprogram 

But I still can't build tinygo on my mac

llvm7 seems to be installed correctly but tinygo seems to look for it in the wrong folder
```
/*
#cgo CFLAGS: -I/usr/lib/llvm-7/include
#cgo LDFLAGS: -L/usr/lib/llvm-7/lib -lclang
*/
```
I forked the tinygo repo and changed the flags in loader to
```
/*
#cgo CFLAGS: -I/usr/local/opt/llvm7/include
#cgo LDFLAGS: -L/usr/local/opt/llvm7/lib -lclang
*/
```
But building tinygo still results in an error:
```
VZMS-MBP-2013-Sierra:tinygo Gerd$ go install
# github.com/Zambiorix/tinygo/loader
Undefined symbols for architecture x86_64:
  "_clang_Cursor_getArgument", referenced from:
      __cgo_2567d3c94687_Cfunc_clang_Cursor_getArgument in _x002.o
     (maybe you meant: __cgo_2567d3c94687_Cfunc_clang_Cursor_getArgument)
  "_clang_Cursor_getNumArguments", referenced from:
      __cgo_2567d3c94687_Cfunc_clang_Cursor_getNumArguments in _x002.o
     (maybe you meant: __cgo_2567d3c94687_Cfunc_clang_Cursor_getNumArguments)
  "_clang_Type_getSizeOf", referenced from:
      __cgo_2567d3c94687_Cfunc_clang_Type_getSizeOf in _x002.o
     (maybe you meant: __cgo_2567d3c94687_Cfunc_clang_Type_getSizeOf)
  "_clang_createIndex", referenced from:
      __cgo_2567d3c94687_Cfunc_clang_createIndex in _x002.o
     (maybe you meant: __cgo_2567d3c94687_Cfunc_clang_createIndex)
  "_clang_disposeIndex", referenced from:
      __cgo_2567d3c94687_Cfunc_clang_disposeIndex in _x002.o
     (maybe you meant: __cgo_2567d3c94687_Cfunc_clang_disposeIndex)
  "_clang_disposeString", referenced from:
      __cgo_2567d3c94687_Cfunc_clang_disposeString in _x002.o
     (maybe you meant: __cgo_2567d3c94687_Cfunc_clang_disposeString)
  "_clang_disposeTranslationUnit", referenced from:
      __cgo_2567d3c94687_Cfunc_clang_disposeTranslationUnit in _x002.o
     (maybe you meant: __cgo_2567d3c94687_Cfunc_clang_disposeTranslationUnit)
  "_clang_getArgType", referenced from:
      __cgo_2567d3c94687_Cfunc_clang_getArgType in _x002.o
     (maybe you meant: __cgo_2567d3c94687_Cfunc_clang_getArgType)
  "_clang_getCString", referenced from:
      __cgo_2567d3c94687_Cfunc_clang_getCString in _x002.o
     (maybe you meant: __cgo_2567d3c94687_Cfunc_clang_getCString)
  "_clang_getCursorKind", referenced from:
      __cgo_2567d3c94687_Cfunc_clang_getCursorKind in _x002.o
     (maybe you meant: __cgo_2567d3c94687_Cfunc_clang_getCursorKind)
  "_clang_getCursorResultType", referenced from:
      __cgo_2567d3c94687_Cfunc_clang_getCursorResultType in _x002.o
     (maybe you meant: __cgo_2567d3c94687_Cfunc_clang_getCursorResultType)
  "_clang_getCursorSpelling", referenced from:
      __cgo_2567d3c94687_Cfunc_clang_getCursorSpelling in _x002.o
     (maybe you meant: __cgo_2567d3c94687_Cfunc_clang_getCursorSpelling)
  "_clang_getCursorType", referenced from:
      __cgo_2567d3c94687_Cfunc_clang_getCursorType in _x002.o
     (maybe you meant: __cgo_2567d3c94687_Cfunc_clang_getCursorType)
  "_clang_getNumDiagnostics", referenced from:
      __cgo_2567d3c94687_Cfunc_clang_getNumDiagnostics in _x002.o
     (maybe you meant: __cgo_2567d3c94687_Cfunc_clang_getNumDiagnostics)
  "_clang_getPointeeType", referenced from:
      __cgo_2567d3c94687_Cfunc_clang_getPointeeType in _x002.o
     (maybe you meant: __cgo_2567d3c94687_Cfunc_clang_getPointeeType)
  "_clang_getTranslationUnitCursor", referenced from:
      __cgo_2567d3c94687_Cfunc_clang_getTranslationUnitCursor in _x002.o
     (maybe you meant: __cgo_2567d3c94687_Cfunc_clang_getTranslationUnitCursor)
  "_clang_getTypeSpelling", referenced from:
      __cgo_2567d3c94687_Cfunc_clang_getTypeSpelling in _x002.o
     (maybe you meant: __cgo_2567d3c94687_Cfunc_clang_getTypeSpelling)
  "_clang_getTypedefDeclUnderlyingType", referenced from:
      __cgo_2567d3c94687_Cfunc_clang_getTypedefDeclUnderlyingType in _x002.o
     (maybe you meant: __cgo_2567d3c94687_Cfunc_clang_getTypedefDeclUnderlyingType)
  "_clang_getTypedefName", referenced from:
      __cgo_2567d3c94687_Cfunc_clang_getTypedefName in _x002.o
     (maybe you meant: __cgo_2567d3c94687_Cfunc_clang_getTypedefName)
  "_clang_isFunctionTypeVariadic", referenced from:
      __cgo_2567d3c94687_Cfunc_clang_isFunctionTypeVariadic in _x002.o
     (maybe you meant: __cgo_2567d3c94687_Cfunc_clang_isFunctionTypeVariadic)
  "_clang_parseTranslationUnit2", referenced from:
      __cgo_2567d3c94687_Cfunc_clang_parseTranslationUnit2 in _x002.o
     (maybe you meant: __cgo_2567d3c94687_Cfunc_clang_parseTranslationUnit2)
  "_clang_visitChildren", referenced from:
      __cgo_2567d3c94687_Cfunc_clang_visitChildren in _x002.o
     (maybe you meant: __cgo_2567d3c94687_Cfunc_clang_visitChildren)
ld: symbol(s) not found for architecture x86_64
clang-7: error: linker command failed with exit code 1 (use -v to see invocation)
```

any idea how I can build tinygo on my mac?

aykevl, 2019-02-24:
I'm not exactly sure what goes wrong here. Maybe `-lclang` links in `libclang.a` instead of `libclang.so`?

Zambiorix, 2019-02-24:
I can't find a `libclang.so` or `libclang.a` on my system, only  `libclang.dylib` ...

aykevl, 2019-02-24:
Ah, yes, OS X uses `.dylib` instead of `.so`.
I have very little experience with OS X and I haven't seen that error before so I don't really know how to fix that error, to be honest.

Zambiorix, 2019-02-24:
Ok, I fixed it ... it had nothing to do with the `.dylib` after all

I changed `loader/libclang_config.go` to:
```
/*
#cgo CFLAGS: -I/usr/local/opt/llvm7/include
#cgo LDFLAGS: -L/usr/local/opt/llvm7/lib -Wl,-rpath,/usr/local/opt/llvm7/lib -lclang
*/
import "C"
```

Then created an extra symlink:
```
cd /usr/local/opt/
ln -s ../Cellar/llvm7/7.0.1 llvm
```

Then installed tinygo
```
go install
```

That returned without any errors .. :)

But  trying to build a wasm project:
```
tinygo build  -o wasm.wasm -target wasm wasm.go
```

returns the error:
```
error: failed to link /var/folders/zh/mc43bk2n5nb0cbn8_6y6gr640000gp/T/tinygo784565872/main: exec: "wasm-ld-7": executable file not found in $PATH
```

any clues?

aykevl, 2019-02-24:
Great! Almost there.
Apparently `wasm-ld-7` wasn't found, which is needed to link the wasm file. It should be installed but wasn't reachable, so adding it to `$PATH` should fix it.

Zambiorix, 2019-02-24:
I finally have it working for OS X High Sierra with a non-forked tinygo (not sure about other versions)

What I did:

1. disable sip first (https://www.imore.com/how-turn-system-integrity-protection-macos)
2. `sudo ln -s /usr/local/Cellar/llvm7/7.0.1 /usr/lib/llvm-7`
3. enable sip again
4. `ln -s /usr/local/Cellar/llvm7/7.0.1/bin/wasm-ld /usr/local/bin/wasm-ld-7`
5. `go get -u github.com/tinygo-org/tinygo`

I still got a warning for the go get command:

```
# github.com/tinygo-org/tinygo
ld: warning: directory not found for option '-L/usr/local/opt/llvm/lib'
```

But it had no impact on building for wasm (not sure about other target). And I was uncomfortable about symlinking `llvm ` in my `/usr/local/opt` path.

`tinygo build  -o wasm.wasm -target wasm wasm.go`
worked like a charm now ... yeay!

aykevl, 2019-02-24:
The `dev` branch has much better support for OSX and should not need such hacks. We will likely do a new release in the next few days, switching over the master (default) branch.

The warning is totally harmless and should also be gone with the `dev` branch.

Zambiorix, 2019-03-07:
I'm not sure what "support for OSX" exactly  means. But reverting the changes I made (symlinks to llvm etc) and reinstalling tinygo resulted in the same issues with llvm.
I had to do step 4 in my previous message to get it working again.
The example (wasm_exec.js changes) made work on Safari, that is true.

Is there a reason why the default wasm_exec.js from  [](https://github.com/golang/go/tree/master/misc/wasm)  can't be used?

I know there are changes, but is there no other way to work around them and keep using the default wasm_exec ?

aykevl, 2019-03-08:
> I'm not sure what "support for OSX" exactly means. But reverting the changes I made (symlinks to llvm etc) and reinstalling tinygo resulted in the same issues with llvm.

That's weird, in testing it works just fine. Have you updated the github.com/tinygo-org/go-llvm package, or even better did you run `dep ensure`?

> Is there a reason why the default wasm_exec.js from golang/go can't be used?

Yes, both compilers produce code that is incompatible from each other. And to be honest, the code that the main Go compiler produces is rather odd and inefficient so I would rather not want to try and match it.

Zambiorix, 2019-03-08:
Yes, I have even deleted tinygo + go-llvm from src, cleaned my pkg folder and deleted the tinygo bin.

I have tested it again, and still the same. When installing I get 2 warnings about missing llvm lib
```
ld: warning: directory not found for option '-L/usr/local/opt/llvm/lib'
```

When using tinygo, I get the complaint that `wasm-ld-7` is missing, adding the symlink and it works again.

I have installed llvm7 via the link mentioned above: https://github.com/tinygo-org/homebrew-tools
Maybe that is the problem?

deadprogram, 2019-04-04:
Hi @Zambiorix the `dev` branch is now based on llvm-8 and also has much improved macOS support. Perhaps you can give that a try?

motecshine, 2019-04-20:
@deadprogram  using dev branch, has some error

```
CGO_CPPFLAGS="-I/Users/zhuhaifeng/Desktop/workspace/gocode/src/github.com/tinygo.org/tinygo/llvm/include -I/Users/zhuhaifeng/Desktop/workspace/gocode/src/github.com/tinygo.org/tinygo/llvm-build/include -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -I/Users/zhuhaifeng/Desktop/workspace/gocode/src/github.com/tinygo.org/tinygo/llvm/tools/clang/include -I/Users/zhuhaifeng/Desktop/workspace/gocode/src/github.com/tinygo.org/tinygo/llvm/tools/lld/include" CGO_CXXFLAGS="-std=c++11" CGO_LDFLAGS="-Lllvm-build/lib  /Users/zhuhaifeng/Desktop/workspace/gocode/src/github.com/tinygo.org/tinygo/llvm-build/lib/libclang.a -lclangAnalysis -lclangARCMigrate -lclangAST -lclangASTMatchers -lclangBasic -lclangCodeGen -lclangCrossTU -lclangDriver -lclangDynamicASTMatchers -lclangEdit -lclangFormat -lclangFrontend -lclangFrontendTool -lclangHandleCXX -lclangHandleLLVM -lclangIndex -lclangLex -lclangParse -lclangRewrite -lclangRewriteFrontend -lclangSema -lclangSerialization -lclangStaticAnalyzerCheckers -lclangStaticAnalyzerCore -lclangStaticAnalyzerFrontend -lclangTooling -lclangToolingASTDiff -lclangToolingCore -lclangToolingInclusions -lclangToolingRefactor  -lstdc++  -llldCOFF -llldCommon -llldCore -llldDriver -llldELF -llldMachO -llldMinGW -llldReaderWriter -llldWasm -llldYAML  -L/Users/zhuhaifeng/Desktop/workspace/gocode/src/github.com/tinygo.org/tinygo/llvm-build/lib -Wl,-search_paths_first -Wl,-headerpad_max_install_names -lLLVMOption -lLLVMMCJIT -lLLVMLTO -lLLVMPasses -lLLVMObjCARCOpts -lLLVMInterpreter -lLLVMExecutionEngine -lLLVMRuntimeDyld -lLLVMDebugInfoDWARF -lLLVMCoroutines -lLLVMipo -lLLVMInstrumentation -lLLVMVectorize -lLLVMLinker -lLLVMIRReader -lLLVMAsmParser -lLLVMAVRDisassembler -lLLVMAVRCodeGen -lLLVMAVRAsmParser -lLLVMAVRDesc -lLLVMAVRInfo -lLLVMAVRAsmPrinter -lLLVMWebAssemblyDisassembler -lLLVMWebAssemblyCodeGen -lLLVMWebAssemblyDesc -lLLVMWebAssemblyAsmPrinter -lLLVMWebAssemblyAsmParser -lLLVMWebAssemblyInfo -lLLVMAArch64Disassembler -lLLVMAArch64CodeGen -lLLVMAArch64AsmParser -lLLVMAArch64Desc -lLLVMAArch64Info -lLLVMAArch64AsmPrinter -lLLVMAArch64Utils -lLLVMARMDisassembler -lLLVMARMCodeGen -lLLVMARMAsmParser -lLLVMARMDesc -lLLVMARMInfo -lLLVMARMAsmPrinter -lLLVMARMUtils -lLLVMX86Disassembler -lLLVMX86AsmParser -lLLVMX86CodeGen -lLLVMGlobalISel -lLLVMSelectionDAG -lLLVMAsmPrinter -lLLVMCodeGen -lLLVMTarget -lLLVMScalarOpts -lLLVMInstCombine -lLLVMAggressiveInstCombine -lLLVMTransformUtils -lLLVMBitWriter -lLLVMAnalysis -lLLVMProfileData -lLLVMX86Desc -lLLVMObject -lLLVMMCParser -lLLVMBitReader -lLLVMCore -lLLVMMCDisassembler -lLLVMX86Info -lLLVMX86AsmPrinter -lLLVMMC -lLLVMDebugInfoCodeView -lLLVMDebugInfoMSF -lLLVMBinaryFormat -lLLVMX86Utils -lLLVMSupport -lLLVMDemangle -lz -lcurses -lm -lxml2" go build -o build/tinygo -tags byollvm .
# tinygo.org/x/go-llvm
InstrumentationBindings.cpp:31:19: error: use of undeclared identifier 'createThreadSanitizerPass'; did you mean 'createHWAddressSanitizerPass'?
/Users/zhuhaifeng/Desktop/workspace/gocode/src/github.com/tinygo.org/tinygo/llvm/include/llvm/Transforms/Instrumentation.h:155:15: note: 'createHWAddressSanitizerPass' declared here
InstrumentationBindings.cpp:35:19: error: use of undeclared identifier 'createMemorySanitizerPass'
make: *** [build/tinygo] Error 2
```

aykevl, 2019-04-20:
Have you tried running `dep ensure --vendor-only` in the tinygo directory?

motecshine, 2019-04-20:
@aykevl sane error

```
# runtime/cgo
ld: warning: directory not found for option '-Lllvm-build/lib'
# os/user
ld: warning: directory not found for option '-Lllvm-build/lib'
# github.com/tinygo-org/tinygo/loader
ld: warning: directory not found for option '-Lllvm-build/lib'
# tinygo.org/x/go-llvm
InstrumentationBindings.cpp:31:19: error: use of undeclared identifier 'createThreadSanitizerPass'; did you mean 'createHWAddressSanitizerPass'?
/Users/zhuhaifeng/Desktop/workspace/gocode/src/github.com/tinygo.org/tinygo/llvm/include/llvm/Transforms/Instrumentation.h:155:15: note: 'createHWAddressSanitizerPass' declared here
InstrumentationBindings.cpp:35:19: error: use of undeclared identifier 'createMemorySanitizerPass'
make: *** [build/tinygo] Error 2
```

motecshine, 2019-04-20:
Reinstalling solved my problem.

```
brew install llvm  #(it is 8.0.0 version)
cd $GOPATH/src/tinygo.org/x  && make VERSION=8.0.0
export GO111MODULE=on
go get -u -v github.com/tinygo-org/tinygo@dev  
```

aykevl, 2019-04-20:
Yes, you need to have LLVM 8 installed when using the dev branch.
Is it working now?

motecshine, 2019-04-20:
@aykevl  work fine

deadprogram, 2019-04-21:
The release v0.5.0 now supports this, so closing. Thanks everyone who helped with this!

closed 2019-04-21 by deadprogram