tinygo-org/tinygo #85: UART not working on "reel board"

deadprogram, 2018-11-19:
I am trying to add support for the "reel board" which is based on the nrf52840. I have been successful as adding the GPIO, however cannot get anything to happen with either TX or RX from my host. I have tried using both the "serial" and "echo" examples.

This is the board: 
https://docs.zephyrproject.org/latest/boards/arm/reel_board/doc/reel_board.html

This is my WIP:
https://github.com/hybridgroup/tinygo/commit/5dc355e459f97abf872c8a1f884f3c2dcd8829bd

Any thoughts?

aykevl, 2018-11-19:
Nothing seems obviously wrong when looking at the code.
You can take a look at the UART section in the datasheet, maybe there's something different from the other two chips (nrf51, nrf52832): http://infocenter.nordicsemi.com/pdf/nRF52840_PS_v1.0.pdf
I don't have the hardware itself so it's hard for me to debug it.

deadprogram, 2018-11-20:
Understood. Any ideas on how I might go about debugging it myself?

aykevl, 2018-11-20:
Things you can try:

* If GDB is working, you can see whether it is doing what you expect: replace 'flash' with 'gdb' in the `tinygo` command. Stop the currently running program with `Ctrl-C`. If it sleeps, it should be at the `for !rtc_wakeup {` line. The regular `bt` command works for the most part so you can see a backtrace. If it crashed, it _might_ point out the exact location where it did.
* If you don't have GDB, you can simply try to blink an LED every time you write something to the UART, to make sure the program didn't crash.
* If you have one, you can hang a logic analyzer or scope directly on the uart tx pin. If not, try connecting an external UART-to-USB converter. Maybe it _does_ send data but something else doesn't work properly.
* You can also try running something else on the device that writes to the UART, to make really sure everything works fine and it is indeed `tinygo` that is to blame.

Hope this helps!

deadprogram, 2018-11-20:
So far I have tried "blink an LED every time you write something to the UART, to make sure the program didn't crash." and have discovered that the program in fact DID crash.

Now I am going to try "running something else on the device that writes to the UART, to make really sure everything works fine and it is indeed tinygo that is to blame."

deadprogram, 2018-11-20:
I was able to run the UART sample code provided with Zephyr so it is something we are doing wrong or not doing, I suppose.

aykevl, 2018-11-21:
Strange. It looks like `EVENTS_TXDRDY` is never sent.
Do you maybe still get a single character (the first), or nothing at all?

deadprogram, 2018-11-21:
Seems like nothing ever arrives, but hard to tell because by the time it connects the MCU has already gotten into a lockup state.

aykevl, 2018-11-21:
Sleep must be working because blinking an LED works. You can try sleeping on startup before sending the first char, for example `time.Sleep(10 * time.Second)` so you have time to connect a serial device.

aykevl, 2018-11-21:
This is even more reason to make stdout more configurable, to support ARM semihosting for cases like these.

deadprogram, 2018-11-21:
Put in the delay, and was able to connect via serial before my LED blink, but did not receive any data.

deadprogram, 2018-11-21:
This is the DTS file that descibes the board, BTW: https://github.com/zephyrproject-rtos/zephyr/blob/f8cbf7f92a8c68817f192e960e49be3c20f80e1f/boards/arm/reel_board/reel_board.dts

deadprogram, 2018-11-21:
This is the info returned by the pyocd flasher, in case that is of any use:

```
Using runner: pyocd
Flashing Target Device
INFO:board:Target type is nrf52
INFO:root:DP IDR = 0x2ba01477
INFO:root:AP#0 IDR = 0x24770011
INFO:root:AP#1 IDR = 0x02880000
INFO:root:AP#0 ROM table #0 @ 0xe00ff000 (designer=244 part=008)
INFO:root:[0]<e000e000:SCS-M4 class=14 designer=43b part=00c>
INFO:root:[1]<e0001000:DWT class=14 designer=43b part=002>
INFO:root:[2]<e0002000:FPB class=14 designer=43b part=003>
INFO:root:[3]<e0000000:ITM class=14 designer=43b part=001>
INFO:root:[4]<e0040000:TPIU-M4 class=9 designer=43b part=9a1 devtype=11 archid=0000 devid=0:0:ca1>
INFO:root:[5]<e0041000:ETM-M4 class=9 designer=43b part=925 devtype=13 archid=0000 devid=0:0:0>
INFO:root:CPU core is Cortex-M4 r0p1
INFO:root:FPU present
INFO:root:4 hardware watchpoints
INFO:root:6 hardware breakpoints, 4 literal comparators
[====================] 100%
INFO:root:Programmed 48676 bytes (12 pages) at 11.20 kB/s
[100%] Built target flash
```

aykevl, 2018-11-21:
Looks like flow control is missing. That may explain why you're not receiving anything.
However, that does not explain why sending chars does not work, the sender should be oblivious to whether the receiver is ready without flow control.

deadprogram, 2018-11-21:
I thought hardware flow control was disabled by default, at least according to the datasheet. That said, you are certainly correct!

deadprogram, 2018-11-27:
Acquired one of the nrf52840 Dev kits from Nordic and confirmed the same behavior.

aykevl, 2018-11-27:
Yes, hardware flow control is disabled by default on the nef52840. But possibly not on the UART-USB chip (probably the debugger chip). You might want to check which pins are related to flow control and pull those high, just to be sure.

@glennrub do you maybe have an idea what's going on here? Is there anything special about the nrf52840 UART?

deadprogram, 2018-11-28:
OK thanks to @jarz-nordic assist, the problem has been found.

@aykevl you were correct that is an error in the generated wrapper. Seems like extra padding is being added after a `struct` cluster.

For example, this extra padding should not be there:
https://gist.github.com/deadprogram/4f7bff995d5758340ce1da9fe03649c4#file-nrf52840-go-L503

deadprogram, 2018-11-28:
Perhaps https://github.com/aykevl/tinygo/blob/master/tools/gen-device-svd.py#L328 is not setting the next address correctly?

aykevl, 2018-11-28:
I'll have to look into it, maybe this evening. This code used to be relatively clean, but it's a mess at the moment so hard to debug.

deadprogram, 2018-11-28:
Here is my proposed fix: https://github.com/aykevl/tinygo/pull/90

deadprogram, 2018-11-28:
I've tested the regenerated files with nrf51, nrf52, and nrf52840 boards. All work as expected for UART and also still work for GPIO operations.

aykevl, 2018-12-01:
I assume this bug has been fixed by #90. Please reopen if it isn't.

closed 2018-12-01 by aykevl