tinygo-org/tinygo #120: Cryptic compiler error on using non-int parameters in make

kyegupov, 2019-01-04:
The following code compiles by mainline Go:

    buf := make([]byte, x.methodReturningUint64())

but, when attempting to use that code in TinyGo, it crashes with the following error:

```
Call parameter type does not match function signature!
  %10 = call i64 @"(TypeOfX).methodReturningUint64"(i32 %8, i8* %9, i8* undef), !dbg !24845
 i32  call void @runtime.sliceBoundsCheckMake(i64 %10, i64 %10, i8* undef), !dbg !24846
```

aykevl, 2019-01-04:
On which target is this an issue? I'm asking because this case should be tested in testdata/slice.go, and I can't reproduce on AVR either.

kyegupov, 2019-01-08:
Sorry, the target is WebAssembly.

aykevl, 2019-01-09:
I managed to reproduce this, with the following code:

```go

var x []byte

func main() {
	buf := make([]byte, foo())
	x = buf
}

func foo() uint64 {
	return 3
}
```

Apparently there is a difference between `uint64(3)` and `foo()`. This is why the tests pass but the code here still fails. I'll have to investigate why this is.

aykevl, 2019-01-09:
And this is a bug on all non-64-bit targets I just tested (avr, wasm, cortex-m). So it's not limited to WebAssembly.

deadprogram, 2019-04-11:
I think this issue is now fixed in the `dev` branch, is that correct @aykevl ?

aykevl, 2019-04-11:
Yes, it should be fixed.

closed 2019-04-11 by aykevl