tinygo-org/tinygo #124: Sometimes WASM code generates calls to memmove, not runime.memmove

kyegupov, 2019-01-05:
E.g. I have a struct "block" with a struct field "header".
An assignment in the form of `header := block.header` sometimes (when LLVM inlining doesn't kick in) generates a code like this:

    i32.const 120
    call $runtime.alloc
    get_local 5
    i32.const 120
    call $memmove

And that causes also `(import "env" "memmove" (func $memmove (type 1)))` to appear, even though we have perfectly good `$runtime.memmove` function.

aykevl, 2019-01-06:
Interesting. This probably happens in a late stage of the compiler, during WebAssembly codegen. And probably because the value is quite big (120 bytes).
At what optimization level are you running this? The default one, which is `-opt=z`?

I think `runtime.memmove` needs a pragmatic like `//go:export memmove`.

deadprogram, 2019-09-17:
Is this still an issue with the current `dev` branch?

jaddr2line, 2020-02-29:
Is this still an issue with the recent addition of a libc for wasm?

aykevl, 2020-02-29:
No this should be fixed now.

closed 2020-02-29 by aykevl