tinygo-org/tinygo #140: Cannot initialize a top-level func var by applying a function to another top-level func var

kyegupov, 2019-01-18:
Discovered when trying to write a deserializer for structures with "generic" (e.g. interface) fields.

A simple example:
```
package factoryfactory

type ifoo interface{}
type ibar interface{}

type generic struct{ foo ifoo }
type foo struct{ bar ibar }
type bar struct{}

func noop(x ifoo) {}

func ifooFactory(ibarFactory func() ibar) func() ifoo {
	return func() ifoo { return foo{ibarFactory()} }
}

var myIbarFactory = func() ibar { return bar{} }
var myIfooFactory = ifooFactory(myIbarFactory)

func main() {
	var x ifoo = myIfooFactory()
	noop(x) // not unused
	return
}
```
Compilation failure:
```
$ tinygo build -wasm-abi=generic -o ff.wasm factoryfactory.go 
panic: interp: load from a pointer bitcast: &PointerCastValue{Value: &AllocaValue{Type: PointerType(StructType(PointerType(StructType(PointerType(IntegerType(8 bits)), PointerType(FunctionType(PointerType(IntegerType(8 bits))):%runtime._interface: StructType(IntegerType(32 bits), PointerType(IntegerType(8 bits))))))))}, CastType: PointerType(PointerType(IntegerType(8 bits)))}
```

kyegupov, 2019-01-18:
(the workaround is 
```
func myIfooFactory() ifoo {
	return ifooFactory(myIbarFactory)
}
```
)

deadprogram, 2019-06-06:
Just wondering if this was possibly already addressed with all of the recent changes.

aykevl, 2019-06-06:
Looks like it's fixed.

closed 2019-06-06 by aykevl