tinygo-org/tinygo #142: tinygo flash command can only compile HEX files

deadprogram, 2019-01-19:
At the moment, the command `tinygo flash` will automatically assume that the target file to be compiled should be a HEX file, due to this parameter here: https://github.com/aykevl/tinygo/blob/master/main.go#L313

This is a small problem when using the `bossac` command with the UF2 bootloader. `bossac` works as expected only when flashing .BIN files. It will appear to flash the `HEX` file, but does not correctly complete the process of setting the user's code for execution. Hence when the MCU restarts, it returns to the bootloader mode, instead of starting up the user's code.

What I propose is to add a new key named "outputFormat" to the target.json file, which lets you set a desired output format. The default can be `.hex` if not specified.

aykevl, 2019-01-19:
I think it is much easier to detect it based on the presence of the `{hex}` or `{bin}` substrings. That should be trivial to implement as well, and doesn't require an extra option in the JSON file.

aykevl, 2019-01-19:
No, to see whether the "flash" attribute in the JSON file contains `{hex}` or `{bin}`. For example:

        "flash": "bossac -d -i -e -w -v -R --offset=0x2000 {hex}"

vs

        "flash": "bossac -d -i -e -w -v -R --offset=0x2000 {bin}"

deadprogram, 2019-01-20:
That makes a lot more sense. OK I will look at creating PR for that.

deadprogram, 2019-02-25:
This code is now in `dev` branch. Now closing.

closed 2019-02-25 by deadprogram