tinygo-org/tinygo #161: Casting struct to an alias type does not work in call

kyegupov, 2019-01-31:
Repro code:
```
package main

type I interface{}

type A struct {
	// Has to be big enough to avoid inlining
	x int
	y int
	z int
	t int
}

type B A

func foo(a A) {
}

func main() {
	var b B
	foo(A(b))
}
```
Results in:
```
$ tinygo build -wasm-abi=generic -o sc.wasm structcast.go 
Call parameter type does not match function signature!
  %0 = load %structcast.go.B, %structcast.go.B* %b, !dbg !1851
 %structcast.go.A = type { i32, i32, i32, i32 }  call void @structcast.go.foo(%structcast.go.B %0, i8* undef, i8* undef), !dbg !1852
error: verification error after IR construction
```

kyegupov, 2019-01-31:
That's a very interesting issue. The fix looks obvious:
```
--- a/compiler/compiler.go
+++ b/compiler/compiler.go
@@ -1850,7 +1850,7 @@ func (c *Compiler) parseExpr(frame *Frame, expr ssa.Value) (llvm.Value, error) {
                // struct types, as those are actually different in LLVM. Let's just
                // bitcast all struct types for ease of use.
                if _, ok := expr.Type().Underlying().(*types.Struct); ok {
-                       llvmType, err := c.getLLVMType(expr.X.Type())
+                       llvmType, err := c.getLLVMType(expr.Type())
```

but then I ran into this: https://stackoverflow.com/questions/54452189/llvm-invalid-bitcast-when-converting-a-struct-to-identical-struct

kyegupov, 2019-01-31:
The explanation is simple. You cannot bitcast a struct to another struct, you can only bitcast non-aggregate types, like pointers. This creates a problem, since Go's SSA assumes "deref then changetype" order of operations:
```
        t0 = local B (b)
        t1 = *t0
        t2 = changetype A <- B (t1)
        t3 = foo(t2)
        return
```
A realitic solution would be to keep t2 as a pointer, and defer the load operation.

aykevl, 2019-04-06:
Fixed by #258.

PR #258 (2019-04-06): compiler: implement casting named structs and pointers to them
This should finally fix the struct casting bug.

closes #162 
fixes #161
closed 2019-04-06 by aykevl