tinygo-org/tinygo #166: docker: unknown blob.

universam1, 2019-02-04:
When I run the docker pull of this image I consistantly get `docker: unknown blob.`

That is regardless on which operating system I run it. Tested on Windows, Ubuntu, Mint.

Could it be that the image is broken?

aykevl, 2019-02-04:
This is a known problem at the moment, the Docker image appears to be broken.

universam1, 2019-02-04:
thanks for the update!
Locally building works kinda after some fixes.

Cool project! 👍

universam1, 2019-02-05:
@jkaflik @aykevl Not totally relevant to this issue, but following PR makes the Docker build resilient https://github.com/tinygo-org/tinygo/pull/168

universam1, 2019-02-05:
@OrangeTux To reproduce the problem, you need to ensure that your local built of the docker image is wiped away as well. Docker is smart enough to take your built layer instead of pulling it down again, where it fails.

So to reproduce:

```
docker rm -f $(docker ps -a -q)
docker rmi -f $(docker images -q)
docker pull tinygo/tinygo:latest
```

and you will get:
```
ddc4ad35a2f0: Downloading
unknown blob
```

Not sure who `ddc4ad35a2f0` is, but I suspect that is an upstream container from golang that has been deleted. So a new built should fix it

OrangeTux, 2019-02-05:
@universam1 Thanks. I was able to reproduce the problem. I have also some extra info taken from `sudo dockerd -D`

```
DEBU[2019-02-05T15:04:02.785112719+01:00] Calling GET /_ping
DEBU[2019-02-05T15:04:02.785530307+01:00] Calling GET /v1.38/info
WARN[2019-02-05T15:04:02.814910766+01:00] failed to retrieve docker-runc version: unknown output format: runc version 1.0.0-rc5+dev
spec: 1.0.0

DEBU[2019-02-05T15:04:02.844550459+01:00] Calling POST /v1.38/images/create?fromImage=tinygo%2Ftinygo&tag=latest
DEBU[2019-02-05T15:04:02.845023993+01:00] Trying to pull tinygo/tinygo from https://registry-1.docker.io v2
DEBU[2019-02-05T15:04:04.400224442+01:00] Pulling ref from V2 registry: tinygo/tinygo:latest
DEBU[2019-02-05T15:04:04.401976120+01:00] pulling blob "sha256:35fba333ff5209042e8925a73f3cbabf00ba725257bdba38ec3b415e5d7e6cc7"
DEBU[2019-02-05T15:04:04.402293046+01:00] pulling blob "sha256:f0cb1fa13079687d5118e5cd7e3ce3c09dc483fa44f0577eca349de8d76e4e8c"
DEBU[2019-02-05T15:04:04.402810943+01:00] pulling blob "sha256:ab1fc7e4bf9195e554669fafa47f69befe22053d7100f5f7002cb9254a36f37c"
DEBU[2019-02-05T15:04:06.459605209+01:00] Downloaded f0cb1fa13079 to tempfile /var/lib/docker/tmp/GetImageBlob294194024
DEBU[2019-02-05T15:04:06.459683844+01:00] pulling blob "sha256:3d1dd648b5ade2bbdfe77fa94424b0314100b58fb5f6a98486538c2126e08e2f"
DEBU[2019-02-05T15:04:08.111125265+01:00] Downloaded 35fba333ff52 to tempfile /var/lib/docker/tmp/GetImageBlob963469037
DEBU[2019-02-05T15:04:08.111236781+01:00] pulling blob "sha256:866ed2e2b1fed04ab28f668a484a1275394fda33f82777064af66a2ba27290ac"
DEBU[2019-02-05T15:04:11.582568929+01:00] Downloaded ab1fc7e4bf91 to tempfile /var/lib/docker/tmp/GetImageBlob373517223
DEBU[2019-02-05T15:04:11.582728272+01:00] pulling blob "sha256:ddc4ad35a2f0d5b45e24d81fb8c61c717f3a042ca270215dcdaab930aada201f"
DEBU[2019-02-05T15:04:11.587903959+01:00] Applying tar in /var/lib/docker/overlay2/859240cbf406484585551db03a1c7017c70dc456dd7e06247654a2f239dde11f/diff  storage-driver=overlay2
ERRO[2019-02-05T15:04:12.564206237+01:00] Download failed: unknown blob
INFO[2019-02-05T15:04:12.564385459+01:00] Attempting next endpoint for pull after error: unknown blob
DEBU[2019-02-05T15:04:12.629798000+01:00] Cleaning up layer 859240cbf406484585551db03a1c7017c70dc456dd7e06247654a2f239dde11f: Error processing tar file(exit status 1): unexpected EOF
```

OrangeTux, 2019-02-05:
> Not sure who `ddc4ad35a2f0` is, but I suspect that is an upstream container from golang that has been deleted. So a new built should fix it

I'm agreeing with @universam1 . I guess that rebuilding and uploading the image will solve the problem.

universam1, 2019-02-05:
I can confirm as well, the container is fixed now!
Thanks for prompt resolution, good job!

closed 2019-02-05 by universam1