tinygo-org/tinygo #185: convert from string: rune needs implementation

deadprogram, 2019-02-11:
I was trying to do the following:

```go
enc := utf16.Encode([]rune("hello"))
```
When I received the following compilation message:

`todo: convert from string: rune.`

Am I correct to assume this will require implementing 2 additional runtime functions? 

```go
stringToRunes(x _string) (slice struct {
	ptr *rune
	len uintptr
	cap uintptr
})
```

and also

```go
encodeUTF16(x rune) ([4]byte, uintptr)
```

aykevl, 2019-02-11:
This is a duplicate of #69. You will need a single runtime function to convert a string to a slice of runes.
I think it can be written in pure Go:

```go
func stringToRunes(s string) []rune
```
I haven't seen any mention of utf-16 in the spec.

See [Conversions to and from a string type](https://golang.org/ref/spec#Conversions):

> Converting a value of a string type to a slice of runes type yields a slice containing the individual Unicode code points of the string.
>
> ```
> []rune(MyString("白鵬翔"))  // []rune{0x767d, 0x9d6c, 0x7fd4}
> []rune("")                 // []rune{}
> 
> MyRunes("白鵬翔")           // []rune{0x767d, 0x9d6c, 0x7fd4}
> ```

aykevl, 2019-03-09:
I'm going to close this as a duplicate of #69.

closed 2019-03-09 by aykevl