tinygo-org/tinygo #186: Panic when trying to compile a package without main function

azuweyx, 2019-02-11:
# Panic when trying to compile a non-main package

## Reproduction

### Code

```go
package login

//go:export loginByEmail
func loginByEmail(email string) (string, error) {
	return "", nil
}
```

### Command

```sh
tinygo build -target wasm -no-debug -o ../../src/assets/login.wasm ../modules/login/login.go
```
## Expected behaviour

Compiles with no problem

## Result

```sh
panic: interface conversion: ssa.Member is nil, not *ssa.Function

goroutine 1 [running]:
github.com/tinygo-org/tinygo/ir.(*Program).SimpleDCE(0xc000407880)
       /go/src/github.com/tinygo-org/tinygo/ir/passes.go:69 +0xf18
github.com/tinygo-org/tinygo/compiler.(*Compiler).Compile(0xc000164160, 0x7fff058a2f2a, 0x20, 0x1b, 0x0)
       /go/src/github.com/tinygo-org/tinygo/compiler/compiler.go:216 +0x59c
main.Compile(0x7fff058a2f2a, 0x20, 0x7fff058a2f09, 0x20, 0xc000156180, 0xc000173f08, 0xc000173cf0, 0x0, 0x0)
       /go/src/github.com/tinygo-org/tinygo/main.go:87 +0x4f2
main.Build(0x7fff058a2f2a, 0x20, 0x7fff058a2f09, 0x20, 0x7fff058a2f01, 0x4, 0xc00009df08, 0xc00009dd80, 0x426c58)
       /go/src/github.com/tinygo-org/tinygo/main.go:289 +0xe2
main.main()
       /go/src/github.com/tinygo-org/tinygo/main.go:555 +0xb5a
```

## Current solution for the issue

Just include an empty main function

### Code

```
package login

func main() {}

//go:export loginByEmail
func loginByEmail(email string) (string, error) {
	return "", nil
}
```

jbpratt, 2020-11-23:
Seems to be a different panic now (using the code block):
```
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x10 pc=0xc13583]

goroutine 1 [running]:
github.com/tinygo-org/tinygo/compiler.Compile(0x7ffee90d32a3, 0x4, 0x61c27a0, 0xc0002482d0, 0x61c4bb0, 0x0, 0x0, 0x0, 0x0, 0x0, ...)
	/home/jbpratt/tinygo/compiler/compiler.go:237 +0x6e3
github.com/tinygo-org/tinygo/builder.Build(0x7ffee90d32a3, 0x4, 0x7ffee90d329c, 0x6, 0xc0002482d0, 0xc000287ba8, 0x0, 0x0)
	/home/jbpratt/tinygo/builder/build.go:51 +0xa5
main.Build(0x7ffee90d32a3, 0x4, 0x7ffee90d329c, 0x6, 0xc000246000, 0x0, 0x12)
	/home/jbpratt/tinygo/main.go:99 +0xc6
main.main()
	/home/jbpratt/tinygo/main.go:921 +0x20ba
```
```
tinygo version 0.16.0 linux/amd64 (using go version go1.15.5 and LLVM version 10.0.1)
```

aykevl, 2020-12-01:
I can't reproduce this, it gives a proper error message for me:

file `nomain/main.go`:

```go
package login

//go:export loginByEmail
func loginByEmail(email string) (string, error) {
	return "", nil
}
```

```
$ tinygo run ./nomain
error: function main is undeclared in the main package
```

@jbpratt how exactly did you reproduce this?

jbpratt, 2020-12-01:
@aykevl Hmm, I can't seem to reproduce this (I'm still on the same build, I believe) so I may have been wrong in the first place. Sorry about the confusion.

aykevl, 2020-12-02:
In that case, it seems to be fixed.

closed 2020-12-02 by aykevl