tinygo-org/tinygo #187: No available targets for wasm

sevki, 2019-02-12:
I checked out https://tinygo.org/getting-started/using-docker/ and modified the tiny-go script to 
look like this but I'm still getting
```
error: No available targets are compatible with this triple.
```
Steps to reproduce:
```
go get -u -v github.com/tinygo-org/tinygo
wget https://raw.githubusercontent.com/tinygo-org/tinygo/master/src/examples/wasm/wasm.go
tinygo build -o main.wasm -target wasm wasm.go
```

```
uname -a 
Linux hawking 4.18.16-arch1-1-ARCH #1 SMP PREEMPT Sat Oct 20 22:06:45 UTC 2018 x86_64 GNU/Linux
```

aykevl, 2019-02-12:
This means the LLVM you use hasn't enabled WebAssembly.
On which OS/distro are you building this and how did you install LLVM 7?

sevki, 2019-02-12:
archlinux

deadprogram, 2019-02-19:
@sevki I think that the version in the Arch package manager does not include WASM by default. You will likely need to install LLVM from source.

aykevl, 2019-02-19:
Yes, this needs to be added to the FAQ, and probably to the installation guide as well. This question has been asked before and will likely come up in the future again.

sevki, 2019-02-19:
I'm sure I'll be wrong but I think my lld has wasm support 
```
wasm-ld --version
LLD 7.0.1
```

aykevl, 2019-02-20:
The error you gave is emitted by LLVM, not by binutils or ld. It fails to even compile the given package, it doesn't even reach the link step.

> I'm sure I'll be wrong but I think my lld has wasm support
> 
>     wasm-ld --version
>     LLD 7.0.1

LLVM 7 does not have WebAssembly enabled by default, so many distros have it disabled (Debian being an exception, as it turns out). You will need to build it from source if you want to use WebAssembly.

Here is a how-to guide for building TinyGo from source. It assumes you want to do a release build but should work just as well for a regular build (you can install the resulting tarball the normal way if needed).  
https://github.com/tinygo-org/tinygo/blob/master/BUILDING.md

sevki, 2019-02-25:
cool. Close the issue when you see fit

sevki, 2019-02-25:
clicked the wrong thing. my bad

closed 2019-02-25 by sevki

anatol, 2020-03-02:
FYI: Arch Linux added `tinygo` package to the official repos https://www.archlinux.org/packages/community/x86_64/tinygo/ Please check it out and report if you see any issues.

PS thank you for this awesome project!

aykevl, 2020-05-16:
For completeness: now that WebAssembly is an official target in LLVM, it is enabled by default on basically all Linux distros so there is no need for building from source just to support WebAssembly.