tinygo-org/tinygo #190: Windows 10 Compilation Problem

hgabe, 2019-02-19:
# github.com/tinygo-org/tinygo/loader
go\src\github.com\tinygo-org\tinygo\loader\libclang.go:16:10: fatal error: clang-c/Index.h: No such file or directory
 #include <clang-c/Index.h> // if this fails, install libclang-7-dev
          ^~~~~~~~~~~~~~~~~
compilation terminated.
# tinygo.org/x/go-llvm
go\src\tinygo.org\x\go-llvm\analysis.go:17:10: fatal error: llvm-c/Analysis.h: No such file or directory
 #include "llvm-c/Analysis.h" // If you are getting an error here read bindings/go/README.txt
          ^~~~~~~~~~~~~~~~~~~
compilation terminated.

OS: Windows 10 64-bit
Using LLVM 7.0.1 pre-built binaries for windows
and gcc 8.2 I tried to compile once before this error because it needed gcc in path

aykevl, 2019-02-19:
Unfortunately, Windows is not yet supported. MacOS has almost landed in #182 but Windows will be more complicated as it is not in the Unix family. It is something I would like to do, however.

There are two ways you can go about this, until proper support has been added:

 * Build inside WSL, using the Linux guide (use a Debian-based distro for best results). This will most likely work, although you will likely not be able to flash to a MCU from there. You could do this manually afterwards.
 * Manually set the correct paths, I think you can set the CFLAGS and LDFLAGS for that. Or you can create a file `$GOPATH/tinygo.org/x/go-llvm/llvm_windows.go` based on the `*_linux.go` variant with modified paths to find the correct header files. A similar modification will be needed in `$GOPATH/github.com/tinygo-org/tinygo/loader/libclang_config.go` to set the correct paths.  
   If you get this to work, it would be useful to make a PR with the changes so that everyone can benefit from it.

deadprogram, 2019-02-19:
A third option is to use the Docker container, which should run under Windows.

aykevl, 2019-02-20:
I have been [experimenting a bit with Windows support](https://github.com/tinygo-org/tinygo/compare/dev...windows2), but haven't got very far yet. It seems that the package in Chocolatey (used by Travis) does not include the necessary header files, so I'll have to look further.

hgabe, 2019-02-21:
Thank you all for the responses and hopefully windows gets supported when possible, I will try it on my linix pc later today

rsrini7, 2019-02-25:
My-Steps in Windows Subsystem for Linux (WSL): Even No Success.

/usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.22' not found 
	https://askubuntu.com/questions/575505/glibcxx-3-4-20-not-found-how-to-fix-this-error

strings /usr/lib/x86_64-linux-gnu/libstdc++.so.6 | grep GLIBCXX

rsrini@MSI:/mnt/d/Wsl/tinygo0.2.0.linux-amd64/src/examples$ ../../bin/tinygo flash -target arduino ./blinky1
error: cannot find package "errors" in any of:
        /mnt/d/Wsl/tinygo0.2.0.linux-amd64/src/errors (from $GOROOT)
        /usr/local/go/src/errors (from $GOPATH)
        /home/rsrini/go/src/errors
		
--> installed golang-1.11.x

error: failed to build targets/avr.S: exec: "avr-gcc": executable file not found in $PATH
	http://maxembedded.com/2015/06/setting-up-avr-gcc-toolchain-on-linux-and-mac-os-x/

rsrini@MSI:/mnt/d/Wsl$ cd tinygo0.2.0.linux-amd64/src/examples/
rsrini@MSI:/mnt/d/Wsl/tinygo0.2.0.linux-amd64/src/examples$ sudo ../../bin/tinygo flash -port "/dev/ttyS4" -target arduino ./blinky1
[sudo] password for rsrini:
avrdude: ser_open(): can't open device "/dev/ttyS4": Input/output error

avrdude done.  Thank you.

error: failed to flash /tmp/tinygo741530055/main.hex: exit status 1
rsrini@MSI:/mnt/d/Wsl/tinygo0.2.0.linux-amd64/src/examples$

Not working even after tring options mensioned in below WSL issue.
https://github.com/Microsoft/WSL/issues/1929
https://blogs.msdn.microsoft.com/wsl/2017/04/14/serial-support-on-the-windows-subsystem-for-linux/

aykevl, 2019-02-25:
It looks like the code is compiling so you're almost there. Maybe you can flash the binary with avrdude outside of WSL?

rsrini7, 2019-02-25:
Yes.  Working.
Steps i follow:

1. Execute tinygo build upon save file (we can auto-execute upon file change)
../../bin/tinygo build -o ./output/output.hex -target arduino ./blinky1

2. Created ardu-upload.cmd
@echo off
D:\Tools\Xloader\avrdude -CD:\Tools\Xloader\avrdude.conf -v -pm328p -carduino -b115200 -cstk500 -P\\.\COM4 -D -Uflash:w:D:\Wsl\tinygo0.2.0.linux-amd64\src\examples\output\output.hex:i

3. install node "watch-cli" - monitor hex file changes and auto-upload to arduino.
watch -p "output\output.hex" -c "ardu-upload"

deadprogram, 2019-10-24:
We now have experimental Windows 10 support in version 0.9.0 as described here:
https://tinygo.org/getting-started/windows/#windows-native-install-experimental

Now closing, please reopen if needed. Thank you!

closed 2019-10-24 by deadprogram