tinygo-org/tinygo #193: docker: unknown blob

gabbifish, 2019-02-21:
I know that two weeks ago a similar issue was resolved (#166), but it looks like the docker: unknown blob error is happening again when I try to pull the tinygo/tinygo container:

```
$ docker run --rm -v $(pwd):/src tinygo/tinygo tinygo build -o /src/wasm.wasm -target wasm examples/wasm
Unable to find image 'tinygo/tinygo:latest' locally
latest: Pulling from tinygo/tinygo
ab1fc7e4bf91: Downloading [=>                                                 ]  1.371MB/45.34MB
35fba333ff52: Downloading [======>                                            ]   1.49MB/10.78MB
f0cb1fa13079: Downloading 
3d1dd648b5ad: Waiting 
866ed2e2b1fe: Waiting 
ddc4ad35a2f0: Waiting 
ab8da8d8889d: Waiting 
b310355c19b5: Waiting 
96073adc9439: Waiting 
37f6e6384272: Waiting 
3b32a12c0fee: Waiting 
70cf9e0f5717: Waiting 
9ca5d2a80fb1: Waiting 
3b06e4870988: Waiting 
2cdf74966fd7: Waiting 
bcdb0b2296cc: Waiting 
73ca093187dc: Waiting 
93010660490e: Waiting 
dfdcf4eac63c: Waiting 
docker: unknown blob.
```

Is the upstream golang container broken again?

gabbifish, 2019-02-21:
If I build the docker container myself, though, looks like it works fine! Just a heads up it would be worth updating the version in the docker registry. :)

deadprogram, 2019-02-21:
Hi @gabbifish we just need to figure out what the cache problem on Dockerhub is. Perhaps someone more experienced with that can provide some guidance?

sevki, 2019-02-21:
Why don’t you just push no cache build An image and upload that

deadprogram, 2019-02-22:
We've just done exactly that, and it seems to be working now as expected when installed in a clean environment. Please give it another try.

deadprogram, 2019-02-25:
From testing people have mentioned in the Slack channels seems this has been resolved. Now closing please reopen if needed. Thanks everyone!

closed 2019-02-25 by deadprogram