tinygo-org/tinygo #207: Support zero runtime web assembly modules

richardanaya, 2019-02-26:
I noticed when compiling a tinygo web assembly module:

* main gets re-exported as "cwa_main"
* functions without bodies get prefixed with the filename in export 
* io_get_stdout is assumed to exist

it would be nice to have a zero runtime compile option for web assembly that:

* exports main properly as "main"
* exports bodyless functions with just their name without prefix
* does not assume io_get_stdout

abourget, 2019-06-23:
This would be useful to use tinygo in other environments where you don't control the VM, like in Blockchains (thinking of EOS here) for example.

abourget, 2019-06-23:
Perhaps a flag that makes the compile use `1` by default for `stdout` ?  With `tiny build -panic trap`, we can remove `resource_write`, but didn't find any way to get rid of `io_get_stdout`.

abourget, 2019-06-23:
I'd also like to be able to define how the import is declared within the imports section of the wasm. Right now I see:

```
  (import "env" "./wasmthing.go.add" (func $./wasmthing.go.add (type 2)))
```

when defining:

```golang
func add(x, y int) int
```

alone in `wasmthing.go`.  I'd like it to be `add` or something I define, perhaps with `//go:import add` ?

deadprogram, 2019-09-18:
Hello all. I think this latter request has been addressed now via https://github.com/tinygo-org/tinygo/commit/55144ad6085c5c0c28d154589c8a7f2f1568364a

Not sure about the original items as requested.

deadprogram, 2019-12-21:
Closing as I think the requested items now exist. Please reopen or open new issues as needed. Thank you.

closed 2019-12-21 by deadprogram