tinygo-org/tinygo #216: go 1.12 - wasm

Zambiorix, 2019-03-05:
I just updated to 1.12 and when compiling my project with tinygo, I get the following error:

# internal/fmtsort
../../../../../../../../../../../../usr/local/go/src/internal/fmtsort/sort.go:58:19: invalid operation: mapValue (variable of type reflect.Value) has no field or method MapRange

It compiled just fine with 1.11.5 earlier

aykevl, 2019-03-05:
Go 1.12 is not yet supported. I have an in-progress (unfinished) PR that will add support in #211. Some non-trivial changes are needed to add support.

deadprogram, 2019-04-04:
This support has just landed in the `dev` branch, please give it a try @Zambiorix

deadprogram, 2019-04-21:
Release v0.5.0 now supports this, so closing this issue. Thanks!

PR #246 (2019-03-23): Support Go 1.12 for most targets
This PR contains almost all changes of #211, but leaves out testing on Go 1.12 because bare-metal targets are not yet supported.
This PR is still useful, because it supports the updated syscall/js package used in the WebAssembly target. Go 1.11 is still supported.

fixes #216
PR #211 (2019-04-03): all: support Go 1.12

Fixes #216.
closed 2019-04-21 by deadprogram