tinygo-org/tinygo #226: wasm js.NewEventCallback returns error

Zambiorix, 2019-03-08:
The code below compiles fine with tinygo but when running I get the following error:

```
(index):1 Uncaught (in promise) LinkError: WebAssembly Instantiation: Import #5 module="env" function="syscall/js.valueInvoke" error: function import requires a callable
Promise.then (async)
(anonymous) @ (index):45
```

```
js.Global().Get("window").Call("addEventListener", "beforeunload", js.NewEventCallback(0, func(event js.Value) {

}))
```

The code compiles and runs without a problem when using the default go compiler

aykevl, 2019-03-08:
This has not yet been implemented, and will likely be implemented after #211.

deadprogram, 2019-05-27:
Code for this has just landed in the `dev` branch with this PR https://github.com/tinygo-org/tinygo/pull/367

Now losing, please reopen if needed. Thank you!

closed 2019-05-27 by deadprogram