tinygo-org/tinygo #233: Compiler freezes with docker wasm example

dicecko, 2019-03-12:
When trying to compile the docker wasm example the compiler freezes.
when running the command from here https://tinygo.org/getting-started/using-docker/

docker run --rm -v $(pwd):/src tinygo/tinygo tinygo build -o /src/wasm.wasm -target wasm examples/wasm

This also happens when creating an own simple example, which has been tested with the regular Go-wasm compiler.
(in makefile in main folder)
build_tinygo:
	docker run --rm \
	-v `pwd`/src3:/game \
	tinygo/tinygo \
	/bin/sh -c "tinygo build -o /game/game.wasm -target wasm /game/main.go; \
	cp /usr/local/go/misc/wasm/wasm_exec.js /game/wasm_exec.js"

It does give other compile errors like unused imports, but then freezes.

deadprogram, 2019-03-12:
Hi @dicecko TinyGo is currently only compatible with Go 1.11, which we expect to address in an upcoming release. Perhaps you can try with that version of Go for now?

aykevl, 2019-03-12:
The container also contains the Go installation so that shouldn't matter.

Can you post the full compiler output? You mention that it freezes after it prints some errors.

Also:
> cp /usr/local/go/misc/wasm/wasm_exec.js /game/wasm_exec.js

TinyGo has its own wasm_exec.js that is not compatible with the main Go wasm_exec.js. You can find it here: https://github.com/tinygo-org/tinygo/blob/master/targets/wasm_exec.js

dicecko, 2019-03-14:
Thanks for the note about the wasm_exec.js

As for the compiler error. It doesn't give any error it just stops responding.

If there are incompatible libraries in the code or unused packages it will return these errors.

However after fixing these or just trying out the original example from the docker it starts compiling and never stops.
No message or anything, but after 25 minutes its still compiling.

aykevl, 2019-03-14:
Weird, one compilation generally takes no more than a few seconds. Can you maybe try some other target, for example `-target pca10040` and see if it still happens? I have seen hangs before in the link stage (`wasm-ld`) but this only happened for WebAssembly and only on Travis (not even in a guest OS on my machine with the same Ubuntu version).

> Could it be that if Go 1.12 is installed on the server it tries to use this instead of Go 1.11 which is included in the docker?

Unlikely, the Docker container should not have access to the host OS unless you explicitly give it. In any case that would not result in a hang but a compiler error.

dicecko, 2019-03-15:
If I run the example with this I get the following error:

$ sudo docker run --rm -v $(pwd):/src tinygo/tinygo tinygo build -o /src/wasm.wasm -target pca10040 examples/wasm

error: failed to make static library /tmp/tinygo-builtins098862579/librt.a: exec: "llvm-ar": executable file not found in $PATH

deadprogram, 2019-03-15:
Hi @dicecko this same issue as https://github.com/tinygo-org/tinygo/issues/235

aykevl, 2019-03-15:
Ok, looks like this is the same `lld` bug as I've seen in Travis as it only happens with wasm. I wonder what triggers it? Maybe it's hardware related?
I'm suspecting this bug, which is fixed in LLVM 8: https://bugs.llvm.org/show_bug.cgi?id=34806

deadprogram, 2019-03-19:
Hello @dicecko have you perhaps had a chance to retry with the latest release?

dicecko, 2019-03-28:
Hi @deadprogram @aykevl , I was without internet for 2 weeks sorry for the late reply.

I just tried it but still freezes.

I did the following:

changed old tinygo folder to tinygo_backup

then did:

$ sudo docker pull tinygo/tinygo

$ cd containerd

$ sudo docker run --rm -v $(pwd):/src tinygo/tinygo tinygo build -o /src/wasm.wasm -target wasm examples/wasm

But then it still freezes.

htop does show this process:

docker-containerd --config /var/run/docker/containerd/containerd.toml --log-level info

but it only uses 0.6% cpu and 2.4% mem so seems to be idle.

Is this issue related to the docker? So if I install it normally on the server it would work?

In the installation it says to use llvm 7 and go 1.11+

But in the previous comment I read that I should use llvm 8? 

And is there any estimate on when the release which supports Go 1.12 will be released?

deadprogram, 2019-04-04:
Hello @dicecko the `dev` branch now has this code. Please give it a try.

deadprogram, 2019-04-04:
@dicecko I just realized that you are using the docker container from dockerhub. That version is the latest released version of TinyGo, but it does not have the updated in the `dev` branch until our next release.

Can you build a local docker container from the `dev` branch to try this out?

deadprogram, 2019-04-21:
Please retry with the new release v0.5.0 and reopen this issue if needed. Thank you.

closed 2019-04-21 by deadprogram