tinygo-org/tinygo #234: LLVM ERROR: Coroutines cannot handle non static allocas yet

NiNJAD3vel0per, 2019-03-12:
**First of all, because normal Go generate super big wasm bundle, I wanted to use tinygo .**

I tried to run and test wasm bundling using the following example(in [https://github.com/golang/go/wiki/WebAssembly](https://github.com/golang/go/wiki/WebAssembly)) of 

**"[](https://github.com/bobcob7/wasm-rotating-cube)"** 

Because I am using MacOS and cannot install llvm and tinygo(I couldn't install),
I ran tinygo inside docker container. However, I got this error!? Did I something wrong? 
or tinygo hasn't support it yet?

**"LLVM ERROR: Coroutines cannot handle non static allocas yet"**

deadprogram, 2019-03-19:
Looks like this feature is not yet fully implemented. Tagging as such.

aykevl, 2019-03-19:
This error is caused by a limitation in the coroutine support of LLVM, not in TinyGo (although TinyGo might be able to fix some of it). I'm surprised to see it, because I thought most of those issues were gone now.
Looking at the code, it also relies on `js.NewCallback` which is not yet supported.

jaddr2line, 2020-03-01:
After #828, there should not be any generated non-static allocas left.

aykevl, 2020-03-01:
That's right. I believe this bug has been fixed now and there is testing in place to avoid future recurrences (see #774). Please comment if there are still instances of it left.

closed 2020-03-01 by aykevl

jaddr2line, 2020-03-01:
The old `coroutines` implementation does still split allocas sometimes I think, so yes there may still be some issues. The new implementation does not.