tinygo-org/tinygo #235: [microbit] Docker biuld fails w/ llvm-ar: executable file not found in $PATH

DazWilkin, 2019-03-14:
Used the generic sample:
```
package main

import (
    "machine"
    "time"
)

func main() {
    led := machine.GPIO{machine.LED}
    led.Configure(machine.GPIOConfig{Mode: machine.GPIO_OUTPUT})
    for {
        led.Low()
        time.Sleep(time.Millisecond * 1000)

        led.High()
        time.Sleep(time.Millisecond * 1000)
    }
}
```
Works for `-target=arduino`:
```
docker run \
--rm \
--interactive \
--tty \
--volume=${GOPATH}/src:/src \
--volume=${GOPATH}/bin:/bin \
tinygo/tinygo tinygo build \
  -o /bin/arduino.hex \
  -target=arduino \
  /src/github.com/DazWilkin/tinygo/arduino/main.go
```
and
```
ls bin
arduino.hex
```
Same code fails for e.g. `-target=bluepill` (another device I have and am interested in using):
```
docker run \
--rm \
--interactive \
--tty \
--volume=${GOPATH}/src:/src \
--volume=${GOPATH}/bin:/bin \
tinygo/tinygo tinygo build \
  -o /bin/bluepill.hex \
  -target=bluepill \
  /src/github.com/DazWilkin/tinygo/arduino/main.go
```
with
```
error: failed to make static library /tmp/tinygo-builtins112190966/librt.a: exec: "llvm-ar": executable file not found in $PATH
```
Fails for a different reason with `-target=microbit`:
```
# /src/github.com/DazWilkin/tinygo/arduino/main.go
../src/github.com/DazWilkin/tinygo/arduino/main.go:9:33: LED not declared by package machine
```
Which I think is because the microbit `machine` package is different:
https://github.com/tinygo-org/tinygo/blob/master/src/machine/board_microbit.go

So tried:
```
package main

import (
    "machine"
)

func main() {
	machine.SetEntireLEDMatrixOn()
}
```
But get the same error as with `-target=bluepill`:
```
docker run \
--rm \
--interactive \
--tty \
--volume=${GOPATH}/src:/src \
--volume=${GOPATH}/bin:/bin \
tinygo/tinygo tinygo build \
  -o /bin/microbit.hex \
  -target=microbit \
  /src/github.com/DazWilkin/tinygo/microbit/main.go
```
and
```
error: failed to make static library /tmp/tinygo-builtins558741771/librt.a: exec: "llvm-ar": executable file not found in $PATH
```

deadprogram, 2019-03-14:
See https://github.com/tinygo-org/tinygo/pull/237

deadprogram, 2019-03-15:
The new release v0.4.1 addresses this issue. Now closing, please reopen if needed. Thank you!

closed 2019-03-15 by deadprogram

DazWilkin, 2019-03-15:
There's currently no Docker image for this release and attempting to build this for myself, fails #239

deadprogram, 2019-03-15:
The latest release should be 0.4.1