tinygo-org/tinygo #236: dockerhub page should provide docker-specific guidance

DazWilkin, 2019-03-14:
https://hub.docker.com/r/tinygo/tinygo

A valid example is provided but it is suggested to use the tinyo binary directly:
```
tinygo flash -target arduino examples/blinky1
```
**NB** I think there's isn't an `examples` directory in the repo!?
**NB** Should it not be `blinky1.go`?

This won't work, of course and may confuse those unfamiliar with Docker|TinyGo.

Recommend:
-- save the example as `blinky1.go` in a working directory (`${PWD}`)
-- from the same directory, run the following Docker command:
```
docker run \
--rm \
--interactive \
--tty \
--volume=${PWD}:/src \
tinygo/tinygo tinygo flash \
  -target=arduino \
  /src/blinky1.go
```

aykevl, 2019-03-15:
You are not really specifying a path on the command line, but a package import path. In the case of examples/blinky1, it resides in src/examples/blinky1 and because TinyGo adds the project root directory to GOPATH this works in general.

For example, this works for me:

    tinygo flash -target=arduino examples/blinky1

It is possible something is broken in the Docker container that causes the GOPATH to not work as well as it should.

aykevl, 2019-03-15:
The Docker Hub page simply shows the README, which was not originally written for Docker.

deadprogram, 2019-03-15:
I agree that the docker page should have something else than just the README. For example this content https://tinygo.org/getting-started/using-docker/ plus a link back to the TinyGo site.

deadprogram, 2019-03-15:
Just updated https://hub.docker.com/r/tinygo/tinygo please take a look, everyone.

DazWilkin, 2019-03-15:
Thanks for doing this.

nit: using docker, we shouldn't assume that the developer has `go get...` or `git clone...` the repo.

@aykevl good point.

deadprogram, 2019-03-15:
OK so are there some immediate changes still needed there?

DazWilkin, 2019-03-15:
I feel guilty that I'm all problems and no solutions, my apologies. I'm unfamiliar with your CI|CD and with how you're publishing to dockerhub so unable to help more.

Previously you had a canonical example shown. I think the guidance could simply be, save this example as a file "blinky1.go" in your working directory and then flash|build it with appropriate command.

johanbrandhorst, 2019-04-12:
The following snippet is how I use the docker image to build wasm files (from https://github.com/johanbrandhorst/wasm-experiments):

```bash
$ docker run --rm -v $(pwd):/go/src/github.com/johanbrandhorst/wasm-experiments --entrypoint /bin/bash tinygo/tinygo:latest -c "\
		cd /go/src/github.com/johanbrandhorst/wasm-experiments && \
		tinygo build -o ./html/test.wasm -target wasm main.go
"
```
Would that serve as a good example? If it could be generalised...

deadprogram, 2019-08-07:
I've updated the docks on the Docker Hub page for the "release" build. Now closing this issue, please reopen or open a new issue on that repo as needed:

https://github.com/tinygo-org/docker

Thanks everyone!

closed 2019-08-07 by deadprogram