tinygo-org/tinygo #239: Unable to build Dockerfile attempting to use release v0.4.1

DazWilkin, 2019-03-15:
I'd had problems installing LLVM correctly on a Linux workstation and so use the Docker image.

Would like to try this now v0.4.1 (#235) but there's no accompanying image.

Self-building the image appears to not work for me :-(

I'm unable to `go get -u github.com/tinygo-org/tinygo` because my local LLVM is not working.

I tried:
```
git clone git@github.com:tinygo-org/tinygo $GOPATH/src/github.com/tinygo-org/tinygo
cd $GOPATH/src/github.com/tinygo-org/tinygo
docker build --tag=tinygo --file=Dockerfile .
```
But this fails when I use this local image:
```
docker run \
--rm --interactive --tty \
--volume=${PWD}/src:/src --volume=${PWD}/bin:/bin \
tinygo tinygo build -o /bin/arduino.hex -target=arduino \
/src/github.com/DazWilkin/tinygo/arduino/main.go
../usr/local/go/src/syscall/js/func.go:74:3: todo: unknown expression: select blocking []
```
but succeeds with the existing deployed image:
```
docker run \
--rm --interactive --tty \
--volume=${PWD}/src:/src --volume=${PWD}/bin:/bin \
tinygo/tinygo tinygo build -o /bin/arduino.hex -target=arduino \
/src/github.com/DazWilkin/tinygo/arduino/main.go
```

Additionally:
- Please consider automatically docker image builds w/ releases

deadprogram, 2019-03-15:
The latest docker image is from the 0.4.1 build.

DazWilkin, 2019-03-15:
Will double-check but:

[image]

DazWilkin, 2019-03-15:
Hmmm :-)

[image]

deadprogram, 2019-03-15:
I agree that we need to get the auto-builds working correctly. They were doing do, but something changed when we switch over to the TinyGo github org, and they have been failing ever since.

I have been doing a manual build/push of Docker image for each release, until we get it working in a automated way.

DazWilkin, 2019-03-15:
My apologies.

The microbit code *does* now compile.

Thank you!

closed 2019-03-15 by DazWilkin

deadprogram, 2019-03-15:
OK, good. No apologies required, you were one of the first people to report, which made it a lot easier to track down. Thanks for helping.

DazWilkin, 2019-03-15:
Thanks for building us such an interesting project!