tinygo-org/tinygo #241: Recommend Dockerfile use `ENTRYPOINT ["tinygo"]`

DazWilkin, 2019-03-17:
The Dockerfile builds an image with `CMD ["tinygo"]` instruction. This provides defaults when running the image but the container will usually be run with parameters. For this reason, it *may* be preferable to use `ENTRYPOINT ["tinygo"]`:

With `CMD`:
```
docker run tinygo/tinygo
No command-line arguments supplied.
TinyGo is a Go compiler for small places.
version: 0.4.1
usage: tinygo command [-printir] [-target=<target>] -o <output> <input>
```
But e.g.:
```
docker run tinygo/tinygo help
docker: Error response from daemon: OCI runtime create failed: container_linux.go:348: starting container process caused "exec: \"help\": executable file not found in $PATH": unknown.
ERRO[0000] error waiting for container: context canceled 
```
This is because `help` overrides the default `tinygo`

So, using `CMD` we must:
```
docker run tinygo/tinygo tinygo help
```
If `ENTRYPOINT ["tinygo"]` were used instead, the following commands would all work:
```
docker run tinygo/tinygo
docker run tinygo/tinygo help
docker run tinygo/tinygo build ...
```
IMO this approach is more elegant in that `docker run ... tinygo/tinygo ...` acts more like the binary on its own `tinygo ...`

**NB** This is a breaking change in that, after switching to `ENTRYPOINT ["tinygo"]`, the existing documentation `...tinygo/tinygo tinygo build` will break because the `tinygo` binary doesn't have a 'tinygo` command.

deadprogram, 2019-03-17:
Please see https://github.com/tinygo-org/tinygo/pull/105 for previous discussion on this.

DazWilkin, 2019-03-17:
Ha! Sorry. It's unclear what problems `ENTRYPOINT` was causing but I'll drop this.

It can be overridden with `--entrypoint=`.

It can be complemented with `CMD` to add defaults.

It reduces side-effects and promotes the use of single-purpose containers.

closed 2019-03-17 by DazWilkin