tinygo-org/tinygo #244: make config make: *** No rule to make target 'config'.  Stop.

toffentoffen, 2019-03-20:
Following the (installation steps)[https://tinygo.org/getting-started/installation/] tells you that:

> It means something is wrong with your LLVM installation. Make sure LLVM 7 is installed (Debian package llvm-7-dev). If it still doesn’t work, you can try running:
```
cd $GOPATH/src/github.com/tinygo-org/go-llvm
make config
```
But there is no make target called config.
I've checked Makefile's history and could fine any trace of a target called config.

Did it ever exist?

aykevl, 2019-03-21:
Did you run `make config` in `$GOPATH/src/github.com/tinygo-org/go-llvm`? It must be run in the directory of this dependency, not in the tinygo directory.

toffentoffen, 2019-03-21:
😂 OMG, sorry! I don't even know how to copy paste

closed 2019-03-21 by toffentoffen