tinygo-org/tinygo #245: SIGSEGV: segmentation violation on version 0.4.x

conejoninja, 2019-03-21:
I'm having a segmentation violation error with every code/example with latest versions (0.4.0 & 0.4.1)

```
tinygo git:(b1744db) tinygo version                                                                                  
tinygo version 0.4.0 linux/amd64
➜  tinygo git:(b1744db) tinygo flash -target=microbit -port=/dev/ttyACM0  github.com/tinygo-org/drivers/examples/vl53l1x
fatal error: unexpected signal during runtime execution
[signal SIGSEGV: segmentation violation code=0x1 addr=0xc0f165d340 pc=0x7fead678d0c8]

runtime stack:
runtime.throw(0x80d0f2, 0x2a)
        /usr/local/go/src/runtime/panic.go:608 +0x72
runtime.sigpanic()
        /usr/local/go/src/runtime/signal_unix.go:374 +0x2f2

goroutine 1 [syscall]:
runtime.cgocall(0x73fd20, 0xc001a2cc40, 0x0)
        /usr/local/go/src/runtime/cgocall.go:128 +0x5e fp=0xc001a2cc10 sp=0xc001a2cbd8 pc=0x41d8de
tinygo.org/x/go-llvm._Cfunc_LLVMGetConstOpcode(0x7feab000c848, 0x7fea00000000)
        _cgo_gotypes.go:5057 +0x49 fp=0xc001a2cc40 sp=0xc001a2cc10 pc=0x5d9839
tinygo.org/x/go-llvm.Value.Opcode.func1(0x7feab000c848, 0x7fea00000000)
        /home/conejo/workspace/go/src/tinygo.org/x/go-llvm/ir.go:862 +0x56 fp=0xc001a2cc78 sp=0xc001a2cc40 pc=0x5fa1d6
tinygo.org/x/go-llvm.Value.Opcode(0x7feab000c848, 0xc0017a0200)
        /home/conejo/workspace/go/src/tinygo.org/x/go-llvm/ir.go:862 +0x2b fp=0xc001a2cc98 sp=0xc001a2cc78 pc=0x5e918b
github.com/tinygo-org/tinygo/interp.(*frame).evalBasicBlock(0xc001a2d4c8, 0x7feab009a9b0, 0x7feab009a690, 0x0, 0x0, 0x0, 0x0, 0xc000a4e498, 0x1, 0x1, ...)
        /home/conejo/workspace/go/src/github.com/tinygo-org/tinygo/interp/frame.go:98 +0x82bb fp=0xc001a2d430 sp=0xc001a2cc98 pc=0x71006b
github.com/tinygo-org/tinygo/interp.(*Eval).function(0xc0018d9080, 0x7feab0017278, 0xc001a2d670, 0x2, 0x2, 0xc001755810, 0x7, 0x0, 0x0, 0x10, ...)
        /home/conejo/workspace/go/src/github.com/tinygo-org/tinygo/interp/interp.go:107 +0x208 fp=0xc001a2d500 sp=0xc001a2d430 pc=0x711308
github.com/tinygo-org/tinygo/interp.(*Eval).Function(0xc0018d9080, 0x7feab0017278, 0xc001a2d670, 0x2, 0x2, 0xc001755810, 0x7, 0x0, 0x0, 0x0, ...)
        /home/conejo/workspace/go/src/github.com/tinygo-org/tinygo/interp/interp.go:90 +0x85 fp=0xc001a2d578 sp=0xc001a2d500 pc=0x7110b5
github.com/tinygo-org/tinygo/interp.Run(0x210d9c0, 0x21035e0, 0x0, 0x0, 0x0)
        /home/conejo/workspace/go/src/github.com/tinygo-org/tinygo/interp/interp.go:77 +0x6a3 fp=0xc001a2d6a0 sp=0xc001a2d578 pc=0x710d73
main.Compile(0x7ffd9b2c7f19, 0x2e, 0x7fcf3c, 0x4, 0xc000158000, 0xc001a2df10, 0xc001a2dce8, 0x0, 0x0)
        /home/conejo/workspace/go/src/github.com/tinygo-org/tinygo/main.go:97 +0x566 fp=0xc001a2dc98 sp=0xc001a2d6a0 pc=0x7305e6
main.Flash(0x7ffd9b2c7f19, 0x2e, 0x7ffd9b2c7efd, 0x8, 0x7ffd9b2c7f0c, 0xc, 0xc0000b9f10, 0xc0000b9d90, 0x426cd8)
        /home/conejo/workspace/go/src/github.com/tinygo-org/tinygo/main.go:337 +0x148 fp=0xc001a2dd28 sp=0xc001a2dc98 pc=0x7334c8
main.main()
        /home/conejo/workspace/go/src/github.com/tinygo-org/tinygo/main.go:591 +0x691 fp=0xc001a2df98 sp=0xc001a2dd28 pc=0x7349b1
runtime.main()
        /usr/local/go/src/runtime/proc.go:201 +0x207 fp=0xc001a2dfe0 sp=0xc001a2df98 pc=0x446e37
runtime.goexit()
        /usr/local/go/src/runtime/asm_amd64.s:1333 +0x1 fp=0xc001a2dfe8 sp=0xc001a2dfe0 pc=0x471b91

goroutine 19 [syscall]:
os/signal.signal_recv(0x0)
        /usr/local/go/src/runtime/sigqueue.go:139 +0x9c
os/signal.loop()
        /usr/local/go/src/os/signal/signal_unix.go:23 +0x22
created by os/signal.init.0
        /usr/local/go/src/os/signal/signal_unix.go:29 +0x41

```

I tried eveything I could think of:
* deleted src/github.com/tinygo-otg/tinygo (and the binary too)
* make clean; make tinygo
* go install

And nothing worked

aykevl, 2019-03-21:
Have you checked src/github.com/tinygo-org/go-llvm? It could be something is wrong with it.

conejoninja, 2019-03-21:
I didn't had it, but go get -u github.com/tinygo-org/go-llvm  and later make config on the folder and same result

aykevl, 2019-03-21:
Just to be sure, can you share the current commit hash and check for local changes in both src/github.com/tinygo-org/go-llvm and src/github.com/tinygo-org/tinygo?
Also, can you tell me your OS and how you installed LLVM?

conejoninja, 2019-03-21:
I think It's fixed now :open_mouth: 
I was using Ubuntu Bionic (kernel 4.15.0-46-generic)
llvm-7-dev was installed (via repositories) but there was an update (as well as some other 700Mb of other packages to upgrade)
upgraded everything, installed tinygo again and ... it worked, sorry for wasting your time

closed 2019-03-21 by conejoninja