tinygo-org/tinygo #252: Undefined reference to `memmove` (time.Location / zoneinfo_unix.go:39)

conejoninja, 2019-03-31:
I'm getting this error trying to compile (for micro:bit if that helps)

```bash
/tmp/tinygo956870840/main.o: In function `time.initLocal':
../../../../../../../../usr/local/go/src/time/zoneinfo_unix.go:39: undefined reference to `memmove'
error: failed to link /tmp/tinygo956870840/main: exit status 1
```
The problem seems to be with the usage of time.UTC (fails too with time.Local and time.FixedZone), if nil is passed, then it fails at runtime (panic: time: missing Location in call to Date
)

The code to reproduce the issue 

```go
// Connects to an MAG3110 I2C magnetometer.
package main

import "time"

func main() {
	dt := time.Date(2019,5,4,3,2,1,0, time.UTC)
	println(dt.Second())
}

```

aykevl, 2019-04-01:
I think there are two issues here:

1. LLVM determined that a `memmove` call is a good idea here. This will need a similar addition as [`memset` here](https://github.com/tinygo-org/tinygo/commit/1bbdab41d2e87a47aafd11471a74a6e02df67099#diff-cdbd36627ae56d846713c2fa18e4a8d8R59), except that memmove is already implemented in the runtime.
2. It looks like the time package is trying to read timezone information, which obviously won't work on a micro:bit. I wonder why it tries to do that, because UTC is in no way system-dependent.

aykevl, 2019-04-01:
@conejoninja can you check whether #253 helps?

deadprogram, 2019-04-03:
The fix for this is now merged into the `dev` branch. Now closing, please re-open if needed.

closed 2019-04-03 by deadprogram