tinygo-org/tinygo #254: Shared object / DLL support

seadra, 2019-04-02:
gc compiler has -buildmode=c-shared.
With LLVM, this should be possible, no?

aykevl, 2019-04-02:
Yes, this is certainly possible. It's just that no one has asked for it yet.
It will need the extra `-buildmode` flag and some extra hooks, like an extra `-static` flag to the linker and  an extra hook to let the compiler emit PIC code instead of static code:

https://github.com/tinygo-org/tinygo/blob/5569cd1b6b3aff8f52052d5ff1028d9bb87fba24/compiler/compiler.go#L105

JasonWeise, 2019-05-28:
This would be a great addition.
I am calling a Go library from Rust and would love to see it compiled with TinyGo instead.

Looking forward to this implementation :)

pontaoski, 2021-01-13:
> Yes, this is certainly possible. It's just that no one has asked for it yet.
> It will need the extra `-buildmode` flag and some extra hooks, like an extra `-static` flag to the linker and  an extra [...]

Is adding this really all that needs to be done? Seems like there would be more to do...

aykevl, 2021-01-13:
It's likely some more plumbing needs to be done but that's the essence of it I think.

riaan-dau, 2022-09-25:
We have an existing C-codebase running bare-metal on 32-bit ARM and would like to make use of the `-buildmode=c-archive` flag to build a static Go library callable from C. Would this be possible if this feature is implemented?

Bmixo, 2022-12-31:
It's almost 2023, how is going to support generate .dll or l.ib

wayneforrest, 2023-10-11:
I am also interested in this feature.

I am using a go .dll / .lib, called from C++; and the DLL is `13MB`, while the C++ code is only `1.5MB.`

31, 2024-01-01:
The Go `c-shared` support doesn't allow `dlclose` (or `FreeLibrary` on Windows), which makes it not ideal for plugins that need to be reloaded. I don't see any reason it'd be easier to support this with TinyGo than Go, but maybe this scenario has more value here? Go issue about this:

* https://github.com/golang/go/issues/11100

turbolent, 2024-11-25:
Is there a simple way to do this manually at the moment. Basically, compile to e.g. `.o` or `.ll`, and link the runtime?

aykevl, 2024-11-30:
Not easily, you'd have to manually initialize the runtime, run package initializers, pass `-scheduler=none` because there is no scheduler, and make sure the functions aren't being called from more than one place.

Maybe I'll add support for this at some point since there seems to be demand for this feature.

soypat, 2026-04-28:
So this is fixed? I think this message is relevant https://gophers.slack.com/archives/CDJD3SUP6/p1775818558605069