tinygo-org/tinygo #262: fmt package don't print negative integers correctly

conejoninja, 2019-04-07:
Looks like it only print positive integers, fmt.Println doesn't do a carriage return either, println does and normal fmt.Println does too.

Example code to reproduce the issue
```go
package main

import (
	"fmt"
)

func main() {
	println("println(-15):", -15)
	println("println(15):", 15)
	println("println(-15.1):", -15.1)
	println("println(15.1):", 15.1)

	fmt.Println("fmt.Println(int32(-15)):", int32(-15))
	fmt.Println("fmt.Println(-15):", -15)
	fmt.Println("fmt.Println(15):", 15)
	fmt.Println("fmt.Println(-15.1):", -15.1)
	fmt.Println("fmt.Println(15.1):", 15.1)

	fmt.Printf("fmt.Printf(-15): %d\r\n", -15)
	fmt.Printf("fmt.Printf(15): %d\r\n", 15)
	fmt.Printf("fmt.Printf(-15.1): %.2f\r\n", -15.1)
	fmt.Printf("fmt.Printf(15.1) %.2f\r\n", 15.1)
}

```

output:
```bash
println(-15): -15
println(15): 15
println(-15.1): -1.510000e+001
println(15.1): +1.510000e+001
fmt.Println(int32(-15)): 4294967281
                                   fmt.Println(-15): 4294967281
                                                               fmt.Println(15): 15
                                                                                  fmt.Println(-15.1): -15.1
  fmt.Println(15.1): 15.1
                         fmt.Printf(-15): 4294967281
fmt.Printf(15): 15
fmt.Printf(-15.1): -15.10
fmt.Printf(15.1) 15.10
```

aykevl, 2019-04-07:
It just so happens that I discovered the underlying issue two days ago and fixed it in 0491407438b40ea9cacd565306ee59d82ad50d06 (part of #260).

conejoninja, 2019-04-07:
I could confirm #260 fixes this issue

closed 2019-04-07 by conejoninja