tinygo-org/tinygo #265: linker: built-in lld elf linker does not respect TINYGOROOT

deadprogram, 2019-04-09:
When using tinygo from the static build on macOS, I discovered that if I tried to compile code from a directory other than that used by the tinygo install, the linker failed:

```shell
tinygo build -o test.uf2 -target circuitplay-express examples/blinky1
tinygo:ld.lld: error: cannot find linker script targets/atsamd21.ld
error: failed to link /var/folders/j1/f8gnf4d91lqgtgcgc47smm_m0000gn/T/tinygo289658339/main: failed to link using built-in ld.lld
```

Digging in a little further, looks like when the executable `tinygo` is being run via symlink ala how homebrew handles `brew install` and the `TINYGOROOT` env var is being used, that the `dir` param for the `Link()` function is not being used in this case: 
https://github.com/tinygo-org/tinygo/blob/dev/linker-builtin.go#L27-L41

Also appears to be ignored for wasm-ld case, but no .ld file needed there, I suppose?

aykevl, 2019-04-09:
Ah, I see. This is actually a bit difficult. It works with the external linker but not the built-in linker because when running the external linker we can set the current working directory. We can't do something like that for the built-in linker (except cwd to the target directory, run the command, and cwd back but that seems fragile to me and prone to concurrency issues).

So my idea would be to _not_ run the external commands in TINYGOROOT and instead make sure that all passed in paths (actually just the linker script) have the correct path. We can do the latter by inserting some special variable (for example, `%tinygoroot%/targets/nrf51.ld`) and replacing them right before linking.

> Also appears to be ignored for wasm-ld case, but no .ld file needed there, I suppose?

That's right, WebAssembly does not need linker scripts. They are only really needed on bare-metal platforms because code and data must be placed on certain addresses.

deadprogram, 2019-04-09:
Having the full correct path to the file in question seems like the safest solution.

aykevl, 2019-04-16:
Turns out the solution is much easier, see #283. The solution I proposed above wouldn't work anyway because some linker scripts include other linker scripts.

PR #283 (2019-04-17): main: fix linker script includes when running outside TINYGOROOT
This commit adds the TinyGo root directory (`TINYGOROOT`) to the linker script `-L` search path, so that linker scripts can be found when running `tinygo` outside of the TinyGo root.

This was already working before when using an external linker by setting the working directory, but this is not possible when using the internal linker. However, by adding the root directory to the linker search path (`-L`), it can now find these linker scripts.

fixes #265
commit 5b34713d (2019-04-17): main: fix linker script includes when running outside TINYGOROOT

This commit adds the TinyGo root directory (`TINYGOROOT`) to the linker
script `-L` search path, so that linker scripts can be found when
running `tinygo` outside of the TinyGo root.

This was already working before when using an external linker by setting
the working directory, but this is not possible when using the internal
linker. However, by adding the root directory to the linker search path
(`-L`), it can now find these linker scripts.

fixes #265
closed 2019-04-20 by aykevl