tinygo-org/tinygo #270: Using image.Point results in    panic: interp: store on a constant

conejoninja, 2019-04-10:
I'm compiling for microbit, to reproduce the issue, add the following to the code

```go
var pt image.Point
```

It results  in 
```bash
panic: interp: store on a constant

goroutine 1 [running]:
github.com/tinygo-org/tinygo/interp.(*LocalValue).Store(0xc0033e6e40, 0x7fa218010a20)
        /home/conejo/workspace/go/src/github.com/tinygo-org/tinygo/interp/values.go:90 +0x204
github.com/tinygo-org/tinygo/interp.(*frame).evalBasicBlock(0xc00455d4b8, 0x7fa21ab7f990, 0x7fa21ab7f350, 0x0, 0x0, 0x0, 0x0, 0xc003196868, 0x1, 0x1, ...)
        /home/conejo/workspace/go/src/github.com/tinygo-org/tinygo/interp/frame.go:113 +0x1aa9
github.com/tinygo-org/tinygo/interp.(*Eval).function(0xc0041450c0, 0x7fa218057618, 0xc00455d660, 0x2, 0x2, 0xc0033e5454, 0x5, 0x0, 0x0, 0x10, ...)
        /home/conejo/workspace/go/src/github.com/tinygo-org/tinygo/interp/interp.go:107 +0x208
github.com/tinygo-org/tinygo/interp.(*Eval).Function(0xc0041450c0, 0x7fa218057618, 0xc00455d660, 0x2, 0x2, 0xc0033e5454, 0x5, 0x0, 0x0, 0x0, ...)
        /home/conejo/workspace/go/src/github.com/tinygo-org/tinygo/interp/interp.go:90 +0x85
github.com/tinygo-org/tinygo/interp.Run(0x26f98b0, 0x26ef4c0, 0x0, 0x0, 0x0)
        /home/conejo/workspace/go/src/github.com/tinygo-org/tinygo/interp/interp.go:77 +0x6a3
main.Compile(0x7ffd78a03ed4, 0x2c, 0x7ff3dc, 0x4, 0xc000156000, 0xc00455df10, 0xc00455dce8, 0x0, 0x0)
        /home/conejo/workspace/go/src/github.com/tinygo-org/tinygo/main.go:90 +0x566
main.Flash(0x7ffd78a03ed4, 0x2c, 0x7ffd78a03eb8, 0x8, 0x7ffd78a03ec7, 0xc, 0xc0000a5f10, 0xc0000a5d90, 0x426cd8)
        /home/conejo/workspace/go/src/github.com/tinygo-org/tinygo/main.go:334 +0x148
main.main()
        /home/conejo/workspace/go/src/github.com/tinygo-org/tinygo/main.go:588 +0x691

```

deadprogram, 2019-10-24:
@conejoninja I'm pretty sure this works now, can you verify?

conejoninja, 2019-10-24:
This compiles now, I close this issue

closed 2019-10-24 by conejoninja