tinygo-org/tinygo #291: Panic during install - Fedora 29 - Go 1.12.3

Munsio, 2019-04-18:
Hi,
As the title describes i have an problem running the `go get` install command of tinygo and i hope someone of you can lend me a hand to fix it.
I tried asking on the slack channel a few times but got no response that's why i'm creating this issue.

My System:

Fedora: 29 - 5.0.5-200.fc29.x86_64
Go: 1.12.3

Packages:
```
avr-binutils.x86_64                               1:2.32-1.fc29                                   
avr-gcc.x86_64                                    1:7.4.0-5.fc29                                        
avr-libc.noarch                                   2.0.0-6.fc29                                           
avrdude.x86_64                                    6.3-13.fc29                         
clang.x86_64                                      7.0.1-6.fc29                                          
clang-devel.x86_64                                7.0.1-6.fc29                                          
clang-libs.x86_64                                 7.0.1-6.fc29                                          
clang-tools-extra.x86_64                          7.0.1-6.fc29           
llvm.x86_64                                       7.0.1-4.fc29                                          
llvm-devel.x86_64                                 7.0.1-4.fc29                                          
llvm-libs.x86_64                                  7.0.1-4.fc29                             
           
```

Error message:

```
# tinygo.org/x/go-llvm
panic: overlapping edits: [8156,8319)->"func() _Ctype_LLVMMetadataRef{ _cgo0 := /*line :297:3*/d.ref; var _cgo1 *_Ctype_char = /*line :298:3*/name; var _cgo2 _Ctype_size_t = _Ctype_size_t(len(t.Name)); var _cgo3 _Ctype_uint64_t = _Ctype_uint64_t(t.SizeInBits); var _cgo4 _Ctype_LLVMDWARFTypeEncoding = _Ctype_LLVMDWARFTypeEncoding(t.Encoding); _cgoCheckPointer(_cgo0); return _Cfunc_LLVMDIBuilderCreateBasicType(_cgo0, _cgo1, _cgo2, _cgo3, _cgo4); }()", [8299,8312)->" /*line :302:3*/_Ctype_LLVMDIFlags /*line :302:16*/"

goroutine 1 [running]:
cmd/internal/edit.(*Buffer).Bytes(0xc000105500, 0xc00000e478, 0x5f171f, 0xe)
	/usr/local/go/src/cmd/internal/edit/edit.go:79 +0x5a4
main.(*Package).writeOutput(0xc0000d8000, 0xc0000c43c0, 0x7ffeccc3fc43, 0x3f)
	/usr/local/go/src/cmd/cgo/out.go:562 +0x38e
main.main()
	/usr/local/go/src/cmd/cgo/main.go:356 +0xceb
```

If someone need more information just ask and i will try to get in touch asap.

aykevl, 2019-04-18:
This sounds like a compiler bug (in the Go compiler, not in TinyGo). I certainly haven't seen this one before.
Can you try running `dep ensure` inside the tinygo source directory, and recompile? It might help, because tinygo.org/x/go-llvm defaults to the llvm8 branch and I see that you're using LLVM 7.

aykevl, 2019-04-18:
Also, I don't remember anything about Fedora nor can I find this issue in the #tinygo Gophers slack.

Munsio, 2019-04-18:
`dep ensure` did the trick. 
after that i used `go install` and the tinygo cmd is available. Thank you so much for help.

After i run:
```
tinygo build -o firmware.hex -size=short -target=arduino myblinky                                                                                                                                                                           
error: No available targets are compatible with this triple.
```

deadprogram, 2019-04-18:
Hi @Munsio the error message `error: No available targets are compatible with this triple.` is coming from LLVM. It indicates that the build of LLVM that you are using does not include the experimental AVR backend support.

One way around this is to build you own custom LLVM. It is easiest to do this using the TinyGo `dev` branch.

Another is to use the release build that will be part of the upcoming 0.5.0 release.

Hope that helps!

Munsio, 2019-04-18:
Thank you @deadprogram i will try using the `dev` branch.
Thank you all for that fast support

Munsio, 2019-04-20:
I changed the distro to Manjaro and tried again the whole process with the dev branch and now its working.
Thanks for pointing me out to this method

closed 2019-04-20 by Munsio