tinygo-org/tinygo #297: Confused about building with llvm now

carolynvs, 2019-04-23:
I think there may be a bug in the latest Makefile on dev? It seems to check for if you have the llvm sources, and doesn't allow for if you have llvm installed by some other means. So when I run `make tinygo` it prompts me to run `make llvm-source; make llvm-build`.

I am not sure if I should have done something when running my make command to tell it about my installation of llvm, or if the target for `build/tinygo` needs to be a bit more flexible.

I have macOS Mojave, and llvm@8 installed via homebrew and was previously able to build tinygo without problems, before the changes to the Makefile.

Any thoughts on how I should be building going forward would be appreciated!

deadprogram, 2019-04-26:
Hi @carolynvs I know from the Slack channel that you were able to get the install working. Can you please confirm that the documentation updates made by @aykevl that are now on the tinygo.org website match what you needed to do? Thank you!

carolynvs, 2019-04-26:
I had no clue that the Makefile was only for people who were using llvm from source. 😀 Yes, that doc makes that more clear that I shouldn't have been using make in the first place. But really I was hoping the fix would be for the Makefile to detect things better, since the other targets are useful, and I was going to add another target to help run the tests for my new feature.

If the case is that I can use make for some targets, but not safely for others, I'd prefer that this issue stay open so that the `build/tinygo` target's logic isn't hard-coded to just work with people who are using llvm from source. Is that an option?

aykevl, 2019-05-03:
Yes, I guess this would be useful. For example, we could do the following:

```sh
# build with LLVM built from source (defaults to LLVM=release_80 at the moment)
make
# build with LLVM from the system
make LLVM=system
# build with LLVM from master
make LLVM=master
# build with LLVM 7
make LLVM=release_70
```

The `LLVM` parameter here would be the name of the LLVM branch we currently use, with a special name "system" indicating that we should build with system-wide installed libs. Would that work?

It isn't really possible to detect this. For example, most of the time I build with system libs, but sometimes I use my self-built LLVM which has assertions enabled (to find bugs in the compiler).

deadprogram, 2019-05-27:
I think this particular issue has now been addressed via both improved docs and also installation scripts/packaging.

Now closing, please reopen if needed. Thank you!

closed 2019-05-27 by deadprogram